ImSwitch/ImSwitch #33: Non constant stack length in a timelapse recording

fpenna, 2021-06-08:
For a given scan setting if I run a timelapse scan the dimension of the different output stacks is not always the same. 

How to reproduce it.
1) I load a scan (the one in the attachment [rs2.zip](https://github.com/kasasxav/ImSwitch/files/6616477/rs2.zip ), that is a 35 nm scan step for a 625 nm length therefore I am expecting a stack of 324 frames. 
2) In the recording I select timelapse scan (the problem stays with any number of frames, but I was generally recording more than 20) without any interval within them (so Freq[s] = 0)
3) once set the right file path and the other needed triggering and laser setting I was just running the measurement
The output stack fluctuates between 324 and 322 frames.

kasasxav, 2021-06-08:
@stafak and I have been trying to solve this bug since forever, so since we didn't manage the plan is to replace time-lapse imaging for scripting, you can try to write you own script for timelapse imaging following the functions in the documentation (https://imswitch.readthedocs.io/en/latest/scripting.html and scripting API reference).
Otherwise, if you can try to investigate where the bug is coming from we may try to debug a bit more, but we have tried many times in the past.

stafak, 2021-06-08:
If I recall correctly this is a synchronization issue; either the recording is starting too late after the first frames have been captured, or ending too early before the last frames are available (I don't remember which it was). A script that records a specific number of frames would solve the issue, but otherwise scripts will probably face similar issues unfortunately.

We have had difficulties with this in the past like Xavi said, but I can take another quick look at the way we handle synchronization next week, to see if I can figure out a solution.

kasasxav, 2021-06-08:
I was thinking maybe with your helping functions (queue and stuff) maybe the sync is more efficient than how we do for the timelaps atm

fpenna, 2021-06-09:
For this measurements I was doing I was putting zero interval between the frames, I can see if introducing a pause/delay eliminate the problem. Does it make sense or you already think that the problem will persist anyway?

stafak, 2021-06-09:
From what I remember the interval doesn't matter, but it wouldn't hurt to double check, so if you have time then sure :) Otherwise I can also look into it further next week.

stafak, 2021-06-17:
So there appears to be a flaw that can result in the last frames of the recording not being saved. I made a rough sketch of how it occurs:

[image]

We might be able to solve this by having the program get the latest frames one final time after the scan has ended. I made a [separate Git branch called "fix-issue-33"](https://github.com/kasasxav/ImSwitch/tree/fix-issue-33) where I applied this fix. If you switch to that branch you can test the fix on your end. Hopefully it works, and there isn't anything else causing this 😅

kasasxav, 2021-06-17:
Makes sense! And the program has no way to know because it's not in a Python buffer, it's in the buffer of the camera !

kasasxav, 2021-06-17:
@stafak I think the best way is to calculate the expected number of frames from the scanning, and then waiting until the controller has them all. I think we can know the exact number of scanning step / pulses in the scheme from the dictionaries in the scanning (usually scanning size / step size)

stafak, 2021-06-17:
Yes, I think that would be better if we can calculate the number of frames!

kasasxav, 2021-06-17:
I think it could be helpful that the ScanDesigner has a parameter totalSteps or something like that. Since each modality will calculate it differently. And in the case of the stage scan I guess is slow_axis_positions*middle_axis_positions*slow_axis_positions. What do you think?

stafak, 2021-06-17:
That's one option, I'll check if it works :)

stafak, 2021-06-21:
Based on our discussion, I attempted to implement a proper fix in the commit referenced above, which I pushed to the dev branch. I'm still not 100% sure that it resolves the issue, but hopefully it does.

kasasxav, 2021-06-24:
There was a small bug where sometimes the scanning ended after the recording, since now the recording stops itself after the expected number of frames are collected.

I fixed it in #7fdfa5f

closed 2021-06-24 by kasasxav