ImSwitch/ImSwitch #38: distribution

untzag, 2021-06-24:
Most scientific Python users prefer to install their non-development software via PyPI or conda-forge. If ImSwitch intends to grow to support non-developer end-users it should be available from these package managers. I'm happy to personally provide guidance on how to set up a pipeline for releasing, if desired.

(I am a JOSS reviewer but this issue is NOT blocking to JOSS publication in my opinion)

stafak, 2021-07-07:
Thanks for your feedback! We have now published ImSwitch on PyPI (handled through a simple pipeline), and for Windows users we are additionally publishing standalone bundles on the releases page on GitHub.

untzag, 2021-07-08:
Thanks @stafak, I'm personally satisfied by what you've done. Lovely to know that ImSwitch is just a `pip install` away now :smile:

closed 2021-07-08 by untzag