ImSwitch/ImSwitch #39: does dependency pinning need to be so agressive?

untzag, 2021-06-24:
Almost all of your dependencies defined in `requirements.txt` are pinned to major.minor.*

Do you really need such aggressive dependency pinning? Unnecessary dependency constraints can be a real hassle for users looking to install ImSwitch alongside other scientific python packages. If you do really depend on such specific package versions, perhaps you could document why under your installation guidelines [1]. This would help users like me understand that I really _need_ to use a dedicated environment for ImSwitch.

[1] https://imswitch.readthedocs.io/en/latest/installation.html

stafak, 2021-06-25:
Hi!

Good point. I imagine that our current approach would be especially problematic if we were to distribute ImSwitch on PyPI/conda-forge, since users would probably expect to be able to simply install the software with a single command and run it without any hassle.

However, I'm not sure what's best practice in the Python world. Should you do something like ">= 1.2, < 2" to pin to a major version (for dependencies that use semantic versioning), just specify a minimum version for each dependency, or avoid pinning completely? And how do you deal with breaking changes in dependencies, in particular in rapidly developing packages like napari – should you just aim to release compatibility patches as quickly as possible?

(Another option for distribution that we have discussed is bundles built with e.g. pyinstaller, instead of publishing on a package repository like PyPI. The pinning would then not be a problem for end-users, but we haven't made a decision on this yet.)

untzag, 2021-06-25:
Hello @stafak :wave:, thanks for your response.

There are definitely multiple valid approaches to dependency pinning. Personally my typical preference is to specify a minimum version only and test aggressively to try and catch breakage from any dependency changes. Github actions allow you to schedule your tests to run once a day [1] which is really useful for catching problems fast. Once you catch a problem you can either change your code, report a bug to upstream, or simply push out a patch excluding the newest versions if you need time to consider your approach.

Bundling is a cool idea that I personally have no experience with. If you're thinking along those lines, you might consider pipx [2].

[1] https://docs.github.com/en/actions/reference/events-that-trigger-workflows#scheduled-events 
[2] https://pypa.github.io/pipx/

stafak, 2021-06-25:
Okay, thank you for your suggestions! We'll give all this some thought and then address it :)

stafak, 2021-07-07:
We have now made the following changes:

- Installs through pip will not have any upper version limit (see [setup.cfg](https://github.com/kasasxav/ImSwitch/blob/master/setup.cfg)). Only one package is pinned, due to an upstream bug that we are awaiting a fix for.
- Tests are re-run on GitHub actions once a day.
- For users who just want to download a known stable version of ImSwitch, we also distribute standalone bundles on the releases page on GitHub (currently only for Windows, but this may change in the future). These bundles include a full Python environment with all packages installed.

untzag, 2021-07-08:
:clap: :clap: very impressive work, @stafak---thank you

closed 2021-07-08 by untzag