ImSwitch/ImSwitch #40: packaging

untzag, 2021-06-24:
ImSwitch does not provide an installable Python package. Please provide `setup.py` or `pyproject.toml` so that users may actually install ImSwitch to their site packages. Running ImSwitch should be done using an entry point [1] so that the software is properly integrated into the operating system.

(This issue must be resolved for me to approve publication in JOSS)

[1] https://packaging.python.org/specifications/entry-points/

untzag, 2021-06-24:
One thing that you will need to do is not store config within your own repository. You should use the operating system's configuration directory for this. I recommend [appdirs](https://github.com/ActiveState/appdirs) which helps you get the config path in an OS-agnostic way.

stafak, 2021-07-07:
Hey @untzag! This is now done -- [pyproject.toml](https://github.com/kasasxav/ImSwitch/blob/master/pyproject.toml) and [setup.cfg](https://github.com/kasasxav/ImSwitch/blob/master/setup.cfg) have been added, and the user files are now placed in a more appropriate location.

untzag, 2021-07-08:
Great, I've installed ImSwitch locally using pip and from PyPI. Very cool. Thanks!

closed 2021-07-08 by untzag