ImSwitch/ImSwitch #42: pickles are not a safe way to store and reload parameters

untzag, 2021-06-24:
Pickles are not designed to be a safe and reliable store of data. They are fragile because they save the entire structure of your objects---even details that you might not care about in representing your data! This means that when you try to unpickle an old set of parameters you will not get an object that's compatible with your current code. Your users will find that their old pickles break when they update your software---even if the contents of the parameters haven't changed.

You only use pickles in one case, to store and recall information from the `slmParameterTree`. Please serialize to a better format such as JSON.

jonatanalvelid, 2021-07-01:
Hi @untzag! Thank you for the comment and suggestion. We have now fixed this issue, and changed pickling to using .json format instead. There are additionally two example .json files included to show what parameters are currently utilized in the code.
Fixed in 15deab9800cfbafc11cc25d59931a4f3cb5c358e.

untzag, 2021-07-08:
Thanks @jonatanalvelid, I like your new approach much more :smile:

closed 2021-07-08 by untzag