ImSwitch/ImSwitch #43: more documentation needed for example scripts / api

untzag, 2021-06-24:
I do not understand how the example scripts work. There is documentation served [1], but it doesn't explain anything.

What is the purpose of `example_import.py` and `example_imported.py`? As far as I can tell they don't interact with ImSwitch at all. Furthermore they don't even work---`python example_import.py` raises an exception `importScript is not defined` (as you would expect).

`example_record.py` does actually interact with ImSwitch, but again the objects you use aren't imported? How am I meant to use this script?

I suspect you are making an assumption that these scripts are run in an environment where some globals are automatically populated into the namespace. As an outsider who isn't aware of those details they're quite confusing.

[1] https://imswitch.readthedocs.io/en/latest/scripting.html

(This issue must be resolved for me to approve publication in JOSS)

stafak, 2021-07-07:
Thank you for the feedback! You are correct in your assumption that the scripts are supposed to be run in a specific environment. I have now tried to clarify this by:

- Adding [a readme in the scripts folder](https://github.com/kasasxav/ImSwitch/blob/master/imswitch/_data/user_defaults/scripts/README.txt) that explains that the scripts must be run in the scripting module in ImSwitch
- Rewording the documentation a little to point to the scripting module, instead of telling the user to check out the scripts folder which could be interpreted as opening it outside ImSwitch

I have also added a short description to each of the example scripts as a comment at the beginning of the code, to make it more clear what they demonstrate. The idea is that the user will understand the basics by looking at the examples.

Let me know if you have any further comments :)

untzag, 2021-07-08:
Thanks @stafak, I somehow totally missed the scripting module.

closed 2021-07-08 by untzag