ImSwitch/ImSwitch #44: community guidelines

untzag, 2021-06-24:
Publication in JOSS requires community guidelines.

>Community guidelines: Are there clear guidelines for third parties wishing to 1) Contribute to the software 2) Report issues or problems with the software 3) Seek support

ImSwitch needs to add contribution guidelines. I'd like to see a contributing page in the documentation at the very least. Given the nature of the software, some more specific guidelines on how people might contribute additional hardware support and control features would be welcome.

(This issue must be resolved for me to approve publication in JOSS)

kasasxav, 2021-06-25:
Thank you for your comment! We have included a contributing page in the documentation and linked it in the readme. Please let us know if you have further feedback regarding this issue.

untzag, 2021-07-08:
Thanks @kasasxav @stafak, looks good to me.

closed 2021-07-08 by untzag