ImSwitch/ImSwitch #45: consider documenting the ImSwitch data file format

untzag, 2021-06-24:
All I can see from your documentation is that ImSwitch saves to HDF5 and that all the metadata is stored. What exactly does the HDF5 file look like? Is it one big array? One array per channel, or per detector? How does it record parameters scanned against? A brief description in the docs would go a long way to helping me know what to expect.

kasasxav, 2021-06-25:
Hi @untzag, thank you for your feedback! We have now included a section in the documentation for explaining the data file and metadata. Please let us know if it is clear enough or if you have further feedback.

untzag, 2021-07-08:
@kasasxav perfect, this is just what I wanted to see. Thanks!

closed 2021-07-08 by untzag