ImSwitch/ImSwitch #57: ImSwitch crashes on Mac Os X

beniroquai, 2021-07-09:
Hey! First of all: Great project and very promising software I would really like to adapt for our @openUC2 system!
I already installed the latest version through pip on windows - works beautifully. 
Today I wanted to start developing customized device adapters using my development environment on Mac. I cloned the current version and did pip. A few issue came up after setting up a fresh environment using conda:

1. The `requirements.txt` was empty
2. The version of `ipython` was not correct - I had to upgrade it 
3. When starting the GUI the following error was thrown:
```
(imswitch) MacBook-Pro:ImSwitch bene$ imswitch
/opt/anaconda3/lib/python3.7/site-packages/visa.py:23: FutureWarning: The visa module provided by PyVISA is being deprecated. You can replace `import visa` by `import pyvisa as visa` to achieve the same effect.

The reason for the deprecation is the possible conflict with the visa package provided by the https://github.com/visa-sdk/visa-python which can result in hard to debug situations.
  FutureWarning,
init master controller
Initializing mock RS232 port
Initializing mock RS232 port
APDred: sigScanInitiate is connected
APDgreen: sigScanInitiate is connected
Initializing Mock TIS
stoplive
Initializing Mock TIS
stoplive
<class 'imswitch.imcontrol.model.SetupInfo.LaserInfo'>
Mock: sending the following to ASRL4::INSTR: L0I1O1
Mock: sending the following to ASRL4::INSTR: L1I1
<class 'imswitch.imcontrol.model.SetupInfo.LaserInfo'>
Mock: sending the following to ASRL4::INSTR: L0I1O1
Mock: sending the following to ASRL4::INSTR: L2I1
Trying to display SLM in monitor 2
2021-07-09 14:56:44.138 python[82478:5234361] -[QNSApplication transformToForegroundApplication]: unrecognized selector sent to instance 0x7fd5566455a0
2021-07-09 14:56:44.163 python[82478:5234361] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[QNSApplication transformToForegroundApplication]: unrecognized selector sent to instance 0x7fd5566455a0'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fff2fc35d07 __exceptionPreprocess + 250
	1   libobjc.A.dylib                     0x00007fff6895a5bf objc_exception_throw + 48
	2   CoreFoundation                      0x00007fff2fcb4c97 -[NSObject(NSObject) __retain_OA] + 0
	3   CoreFoundation                      0x00007fff2fb9a57b ___forwarding___ + 1427
	4   CoreFoundation                      0x00007fff2fb99f58 _CF_forwarding_prep_0 + 120
	5   libwx_osx_cocoau_core-3.1.5.0.0.dyl 0x0000001a4cb5de5e _ZN5wxApp9DoInitGuiEv + 254
	6   libwx_osx_cocoau_core-3.1.5.0.0.dyl 0x0000001a4cbabd7e _ZN5wxApp9OnInitGuiEv + 30
	7   _core.cpython-37m-darwin.so         0x0000001a4c06ca52 _ZN10sipwxPyApp9OnInitGuiEv + 98
	8   libwx_osx_cocoau_core-3.1.5.0.0.dyl 0x0000001a4cbabba5 _ZN5wxApp10InitializeERiPPw + 165
	9   libwx_baseu-3.1.5.0.0.dylib         0x0000001a4d34958b _Z12wxEntryStartRiPPw + 171
	10  _core.cpython-37m-darwin.so         0x0000001a4c06c1a3 _ZN7wxPyApp13_BootstrapAppEv + 499
	11  _core.cpython-37m-darwin.so         0x0000001a4c0703f4 _ZL26meth_wxPyApp__BootstrapAppP7_objectS0_ + 100
	12  python                              0x0000000104209bae _PyMethodDef_RawFastCallKeywords + 350
	13  python                              0x0000000104346de2 call_function + 306
	14  python                              0x0000000104343b92 _PyEval_EvalFrameDefault + 42226
	15  python                              0x000000010433846e _PyEval_EvalCodeWithName + 414
	16  python                              0x00000001042087c7 _PyFunction_FastCallDict + 231
	17  python                              0x000000010428aa61 slot_tp_init + 193
	18  python                              0x0000000104294b51 type_call + 241
	19  python                              0x00000001042094f3 _PyObject_FastCallKeywords + 179
	20  python                              0x0000000104346e75 call_function + 453
	21  python                              0x0000000104343b92 _PyEval_EvalFrameDefault + 42226
	22  python                              0x00000001042092d5 function_code_fastcall + 117
	23  python                              0x0000000104346d67 call_function + 183
	24  python                              0x0000000104343afc _PyEval_EvalFrameDefault + 42076
	25  python                              0x00000001042092d5 function_code_fastcall + 117
	26  python                              0x0000000104346d67 call_function + 183
	27  python                              0x0000000104343afc _PyEval_EvalFrameDefault + 42076
	28  python                              0x00000001042092d5 function_code_fastcall + 117
	29  python                              0x000000010420c702 method_call + 130
	30  python                              0x000000010420a172 PyObject_Call + 130
	31  python                              0x000000010442889b t_bootstrap + 123
	32  python                              0x00000001043afa67 pythread_wrapper + 39
	33  libsystem_pthread.dylib             0x00007fff69d06109 _pthread_start + 148
	34  libsystem_pthread.dylib             0x00007fff69d01b8b thread_start + 15
)
libc++abi.dylib: terminating with uncaught exception of type NSException
Abort trap: 6
(imswitch)MacBook-Pro:ImSwitch bene$ 
```

Any clue what this could mean? Looks a bit more complicated? 
After that, I cannot restart imsswitch from the same terminal.

kasasxav, 2021-07-09:
Hey @beniroquai , that's so cool and exciting that you are trying to make it work for @openUC2 ! I really like the project and definitely will try to help you out.

I have tried to reproduce the error in my mac and realized that it's related to the setup in example_sted.json because of the SLM widget. 

We will try to fix it, but if you want to keep working you can run example_monalisa.json instead, which seems to work. You can also remove the SLM from the list of widgets and the SLM block in the example_sted.json file then it worked for me as well.

For that you have to check the ImSwitch folder in your home directory, then change the file imcontrol_options.json. There you write example_monalisa.json instead of example_sted.json, or in imcontrol_setups/example_sted.json remove what I mentioned.

If you want to try the software out with some of your hardware devices you can try writing a json file there with the list of devices you have and see if it's fine.

Hope it helps! We will continue to look into the issue.

stafak, 2021-07-09:
Regarding the requirements, assuming your pip is up-to-date `pip install -r requirements.txt` should do the same thing as `pip install .` (requirements.txt contains "."; the runtime requirements defined in setup.cfg will be installed automatically).

However, as of the latest commit on master we recommend developers to run `pip install -r requirements-dev.txt` as this will install ImSwitch in editable mode. This should perhaps be clarified in the documentation.

beniroquai, 2021-07-12:
Hey thanks for your reply! It works now! :-)

closed 2021-07-12 by beniroquai