ImSwitch/ImSwitch #59: GUI seems incompatible with retina display on Windows, makes software unusable

uellue, 2021-07-12:
## How to reproduce

* Use most recent release, Windows download .zip on Windows 10 system with high-resolution display (3840x2160, 250 % scale) and follow instructions
* Choose STED example

## What happens

A window without interaction possibility covers the full screen (see screenshot). Other parts of the GUI are behind that window. Keyboard shortcuts have to be used to access the task bar and bring one of the other windows to the foreground.

[image]

If a second monitor with 1920x1080 and 100 % scale is connected, the scale of the main  GUI seems off, at least if it is on the low resolution monitor.

## What should happen

A nice window layout and a way to close or move aside windows without resorting to keyboard shortcuts. :-)

stafak, 2021-07-16:
Hi! :)

I think the issue with the full screen display should be solved now with the latest commits to master. It's supposed to show up on another screen, but when there only is one monitor connected and turned on it falls back to that monitor, which results in a not very nice situation as you noticed. We've made the following changes to the full screen display:

- It will now only show up after clicking "show SLM display" in the SLM widget
- It is now possible to close it by pressing the escape key while it is in focus
- A warning will show up if it is about to show up in front of the main ImSwitch window (which also explains how to close it)

> If a second monitor with 1920x1080 and 100 % scale is connected, the scale of the main GUI seems off, at least if it is on the low resolution monitor.

Unfortunately, this seems to be a Qt problem that there is no easy solution for. I searched the internet for information on this, and the general recommendation seems to be to let Windows handle the scaling instead of Qt, but this leads to bluriness which we believe would not be acceptable in a microscopy program. If a user wants to let Windows handle the scaling anyway, they can do so by [changing the high DPI scaling override setting](https://www.windowscentral.com/how-change-high-dpi-settings-classic-apps-windows-10-april-2018-update) for ImSwitch.exe.

uellue, 2021-07-27:
Hi @stafak, thank you for the explanation and working on a fix! When I tried to test it, I ran into #73, unfortunately.

uellue, 2021-07-27:
@stafak fix confirmed! For some reason the GUI element scaling is much better now, too. 🚀

stafak, 2021-07-28:
Thanks! I'll close this then.

closed 2021-07-28 by stafak