ImSwitch/ImSwitch #73: Examples trigger error in fresh install without connected hardware

uellue, 2021-07-27:
**Bug description**

Selecting the `example_monalisa.json` or `example_sted.json` in a fresh install without connected hardware triggers an error.

**How to reproduce**

1. Install current master of ImSwitch with `pip install -e .` from a Git clone
2. Wipe `Documents\ImSwitch` if present
3. Start `imswitch`
4. Select `example_monalisa.json` or `example_sted.json`

**Expected behavior**

The application works in a demo version or shows a concise error message on what is wrong and what the user can do to fix it.

**Screenshots**

[image]

**Environment**
 - OS: "Windows 10"
 - ImSwitch version: current master at this time
 - How you downloaded ImSwitch: `git clone ...` and `pip install -e`

**Additional context**
Add any other context about the problem here, if applicable.

stafak, 2021-07-27:
Whoops, looks like I missed a thing when I changed the error handling recently. Should be fixed by the commit above (available on master).

uellue, 2021-07-27:
Fix confirmed! 🚀

closed 2021-07-28 by stafak