ImSwitch/ImSwitch #85: Napari Error when displaying camera stream

beniroquai, 2021-10-19:
**Bug description**
As stated earlier, I'm trying to implement a new camera class. **Warning:** I'm not a programmer/a very bad one.. 

**How to reproduce**
Steps to reproduce the behavior:
1. Starting the software using the Allied Vision Camera
2. Starting the live-stream

**Expected behavior**
An image should appear on the Napari image viewer 

**Environment**
 - OS: Windows 10
 - ImSwitch version: Latest forked from this repo, added support for GRBL pcbs and the Allied vision camera
 - Can be found here: https://github.com/beniroquai/imswitch

**Additional context**
The error message I receive is the following (when starting live-stream):

```py
 ** On entry to DGEQRF parameter number  4 had an illegal value
 ** On entry to DGEQRF parameter number  4 had an illegal value
2021-10-19 15:56:03 ERROR [ExceptionHandler] Traceback (most recent call last):
  File "c:\users\uc2\documents\imswitch\imswitch\imcontrol\controller\controllers\ImageController.py", line 59, in update
    self._widget.setImage(detectorName, im)
  File "c:\users\uc2\documents\imswitch\imswitch\imcontrol\view\widgets\ImageWidget.py", line 62, in setImage
    self.imgLayers[name].data = im
  File "C:\Users\UC2\Anaconda3\envs\imswitch\lib\site-packages\napari\layers\image\image.py", line 335, in data
    self._update_dims()
  File "C:\Users\UC2\Anaconda3\envs\imswitch\lib\site-packages\napari\layers\base\base.py", line 536, in _update_dims
    self._transforms = self._transforms.expand_dims(new_axes)
  File "C:\Users\UC2\Anaconda3\envs\imswitch\lib\site-packages\napari\utils\transforms\transforms.py", line 148, in expand_dims
    return TransformChain([tf.expand_dims(axes) for tf in self])
  File "C:\Users\UC2\Anaconda3\envs\imswitch\lib\site-packages\napari\utils\transforms\transforms.py", line 148, in <listcomp>
    return TransformChain([tf.expand_dims(axes) for tf in self])
  File "C:\Users\UC2\Anaconda3\envs\imswitch\lib\site-packages\napari\utils\transforms\transforms.py", line 482, in expand_dims
    n = len(axes) + len(self.scale)
  File "C:\Users\UC2\Anaconda3\envs\imswitch\lib\site-packages\napari\utils\transforms\transforms.py", line 367, in scale
    return decompose_linear_matrix(
  File "C:\Users\UC2\Anaconda3\envs\imswitch\lib\site-packages\napari\utils\transforms\transform_utils.py", line 190, in decompose_linear_matrix
    rotate, tri = scipy.linalg.qr(matrix)
  File "C:\Users\UC2\Anaconda3\envs\imswitch\lib\site-packages\scipy\linalg\decomp_qr.py", line 141, in qr
    qr, tau = safecall(geqrf, "geqrf", a1, lwork=lwork,
  File "C:\Users\UC2\Anaconda3\envs\imswitch\lib\site-packages\scipy\linalg\decomp_qr.py", line 21, in safecall
    raise ValueError("illegal value in %dth argument of internal %s"
ValueError: illegal value in 4th argument of internal geqrf

2021-10-19 15:56:03 DEBUG [AVManager -> WidefieldCamera] suspendlive
2021-10-19 15:56:20 ERROR [ExceptionHandler] Traceback (most recent call last):
  File "C:\Users\UC2\Anaconda3\envs\imswitch\lib\site-packages\vispy\app\backends\_qt.py", line 897, in paintGL
    self._vispy_canvas.events.draw(region=None)
  File "C:\Users\UC2\Anaconda3\envs\imswitch\lib\site-packages\vispy\util\event.py", line 453, in __call__
    self._invoke_callback(cb, event)
  File "C:\Users\UC2\Anaconda3\envs\imswitch\lib\site-packages\vispy\util\event.py", line 471, in _invoke_callback
    _handle_exception(self.ignore_callback_errors,
  File "C:\Users\UC2\Anaconda3\envs\imswitch\lib\site-packages\vispy\util\event.py", line 469, in _invoke_callback
    cb(event)
  File "C:\Users\UC2\Anaconda3\envs\imswitch\lib\site-packages\napari\_qt\qt_viewer.py", line 730, in on_draw
    layer._update_draw(
  File "C:\Users\UC2\Anaconda3\envs\imswitch\lib\site-packages\napari\layers\base\base.py", line 986, in _update_draw
    data_corners = self._transforms[1:].simplified.inverse(corner_pixels)
  File "C:\Users\UC2\Anaconda3\envs\imswitch\lib\site-packages\napari\utils\transforms\transforms.py", line 349, in __call__
    linear_matrix[
ValueError: could not broadcast input array from shape (0,0) into shape (2,2)
```

The error repeats all over and over again. Did you ever encounter the issue? I set a breakpoint before the first frame should show up in Napari, I assume [here](https://github.com/beniroquai/ImSwitch/blob/master/imswitch/imcontrol/controller/controllers/ImageController.py#L56), but the error remains. Any idea? I think it's some thread that receives an empty frame at initialization? But I cannot find it :)

beniroquai, 2021-10-19:
Hey, The framesize is 
```
FRAME_HEIGHT = 1088
FRAME_WIDTH = 1456
```
as I would have expected it. But the very first frame is a `None` array - not sure if this is intended? The allied vision camera code just grabs frames the camera spills out.

Here another screencast:
[image]

kasasxav, 2021-10-19:
Hi @beniroquai , I'm checking the gif but it's a bit hard to zoom in (pixelated) and stop it in the right place, but it seems that there is a logger debug somewhere in the manager and I don't manage to see the print of image Controller, is it true or can you maybe send a screenshot?