ImSwitch/ImSwitch #90: Lantz package error

jacopoabramo, 2021-11-03:
**Bug description**
I'm currently working on my ImSwitch fork to integrate the APIs for the Ximea family cameras. I've tried to install ImSwitch locally and I first got this traceback:

```
2021-11-03 10:35:34 INFO [main] Starting ImSwitch 1.2.1
2021-11-03 10:35:37 ERROR [main] Failed to initialize module imcontrol
2021-11-03 10:35:37 ERROR [main] Traceback (most recent call last):
  File "C:\Users\jacop\AppData\Local\Programs\Python\Python39\lib\site-packages\imswitch\__main__.py", line 50, in main
    view, controller = modulePkg.getMainViewAndController(
  File "C:\Users\jacop\AppData\Local\Programs\Python\Python39\lib\site-packages\imswitch\imcontrol\__init__.py", line 8, in getMainViewAndController
    from .controller import ImConMainController
  File "C:\Users\jacop\AppData\Local\Programs\Python\Python39\lib\site-packages\imswitch\imcontrol\controller\__init__.py", line 1, in <module>
    from .ImConMainController import ImConMainController
  File "C:\Users\jacop\AppData\Local\Programs\Python\Python39\lib\site-packages\imswitch\imcontrol\controller\ImConMainController.py", line 11, in <module>
    from . import controllers
  File "C:\Users\jacop\AppData\Local\Programs\Python\Python39\lib\site-packages\imswitch\imcontrol\controller\controllers\__init__.py", line 7, in <module>
    from .FocusLockController import FocusLockController
  File "C:\Users\jacop\AppData\Local\Programs\Python\Python39\lib\site-packages\imswitch\imcontrol\controller\controllers\FocusLockController.py", line 6, in <module>
    from lantz import Q_
ImportError: cannot import name 'Q_' from 'lantz' (unknown location)
```

I tried to install the lantz package via `pip install lantz`, but this caused the following traceback:

```
2021-11-03 10:39:56 INFO [main] Starting ImSwitch 1.2.1
2021-11-03 10:40:00 ERROR [main] Failed to initialize module imcontrol
2021-11-03 10:40:00 ERROR [main] Traceback (most recent call last):
  File "C:\Users\jacop\AppData\Local\Programs\Python\Python39\lib\site-packages\imswitch\__main__.py", line 50, in main
    view, controller = modulePkg.getMainViewAndController(
  File "C:\Users\jacop\AppData\Local\Programs\Python\Python39\lib\site-packages\imswitch\imcontrol\__init__.py", line 8, in getMainViewAndController
    from .controller import ImConMainController
  File "C:\Users\jacop\AppData\Local\Programs\Python\Python39\lib\site-packages\imswitch\imcontrol\controller\__init__.py", line 1, in <module>
    from .ImConMainController import ImConMainController
  File "C:\Users\jacop\AppData\Local\Programs\Python\Python39\lib\site-packages\imswitch\imcontrol\controller\ImConMainController.py", line 11, in <module>
    from . import controllers
  File "C:\Users\jacop\AppData\Local\Programs\Python\Python39\lib\site-packages\imswitch\imcontrol\controller\controllers\__init__.py", line 7, in <module>
    from .FocusLockController import FocusLockController
  File "C:\Users\jacop\AppData\Local\Programs\Python\Python39\lib\site-packages\imswitch\imcontrol\controller\controllers\FocusLockController.py", line 6, in <module>
    from lantz import Q_
  File "C:\Users\jacop\AppData\Local\Programs\Python\Python39\lib\site-packages\lantz\__init__.py", line 25, in <module>
    from .driver import Driver, Feat, DictFeat, Action, initialize_many, finalize_many
  File "C:\Users\jacop\AppData\Local\Programs\Python\Python39\lib\site-packages\lantz\driver.py", line 19, in <module>
    from .utils.qt import MetaQObject, SuperQObject, QtCore
  File "C:\Users\jacop\AppData\Local\Programs\Python\Python39\lib\site-packages\lantz\utils\qt.py", line 26, in <module>
    raise RuntimeError("Invalid Qt API %r, valid values are: %r, %r, %r" %
RuntimeError: Invalid Qt API 'PyQt5', valid values are: 'pyside', 'pyqt', 'mock'
```

**How to reproduce**
Steps to reproduce the behavior:
1. Install local setup of ImSwitch
2. Call ImSwitch from command line

**Expected behavior**
ImSwitch running with my selected hardware setup.

**Additional information**
This is the content of the json file I'm trying to use to test the camera integration:

```json
{
    "detectors": {
        "HamamatsuTest": {
            "analogChannel": null,
            "digitalLine": 4,
            "managerName": "HamamatsuManager",
            "managerProperties": {
                "cameraListIndex": 0,
                "hamamatsu": {
                    "readout_speed": 3,
                    "trigger_global_exposure": 5,
                    "trigger_active": 2,
                    "trigger_polarity": 2,
                    "exposure_time": 0.01,
                    "trigger_source": 1,
                    "subarray_hpos": 0,
                    "subarray_vpos": 0,
                    "subarray_hsize": 2048,
                    "subarray_vsize": 2048
                }
            },
            "forAcquisition": true
        }
		"XimeaTest": {
            "managerName": "XimeaManager",
            "managerProperties": {
                "cameraListIndex": 0,
                "hamamatsu": {
                    "exposure" : 1000,
                    "offsetX" : 0,
                    "offsetY" : 0,
                    "width" : 1280,
					"height" : 864,
					"trigger_source" : "XI_TRG_OFF"
					"gpi_selector" : "XI_GPI_PORT1"
                }
            },
            "forAcquisition": true
        }
    },
    "rois": {
        "ROI_test": {
            "x": 706,
            "y": 570,
            "w": 839,
            "h": 839
        }
    },
    "availableWidgets": [
        "Settings",
        "View",
        "Recording",
        "Image",
        "BeadRec",
        "FFT",
        "ULenses"
    ]
}
```

**Environment**
 - OS: Windows 10:
 - ImSwitch version: 1.2.1 ([fork](https://github.com/Eggeling-Lab-Microscope-Software/ImSwitch))
 - How you downloaded ImSwitch: standalone bundle

**Additional context**
Changes have not being committed yet to the repo, but for some context:
- I'm integrating the Ximea APIs according to their documentation and with the functionalities required by the DetectorManager;
- I created a mock class for when the Ximea APIs are not installed

kasasxav, 2021-11-03:
We had different problems because `lantz` changed to `lantzdev`, but it's already in [setup.cfg](https://github.com/kasasxav/ImSwitch/blob/master/setup.cfg), so if you do pip install . it should work.
Otherwise, maybe you can try to run `pip install lantzdev` instead?

jacopoabramo, 2021-11-03:
Ok, that fixed it. I must have installed the `lantz` package by mistake. Closing the issue.

closed 2021-11-03 by jacopoabramo