ImSwitch/ImSwitch #92: Feature: Having the Raspberry Pi Camera talking to ImSwitch

beniroquai, 2021-11-03:
Hey, imagine we could have the Raspberry Pi camera (streamed?) inside the ImSwitch environment. With this one could immediately have so many cool open microscopy projects integrated in one go (e.g. UC2, openflexure...). Do you have experience with streaming the data fast? I would be very keen on writing a camera class that can eventually do it. Did you think about it already?

beniroquai, 2021-11-07:
Hey, first prototype is ready. It only works "offline" since I do not yet have a Raspberry Pi to test it with. But the idea is simple. Have a REST-API client-server connection, which handles settings such as exposure and gain on the camera side, as well as the frame handling on the ImSwitch side. Will test it once I have a Pi at hand. 

You can see the changes [here](https://github.com/beniroquai/ImSwitch/commit/703e681e67b090e397d4ccabf553a127d9bebab1) (everything that says something with `*PiCamera`. 

The first version of the server is [here](https://github.com/openUC2/UC2-RESTPiCam) (I tried to make my first installable package out of it - I failed 😉)

kasasxav, 2021-11-09:
Hi @beniroquai, cool suggestion! I actually have no experience with using Raspberry Pi in microscopy but seems like a good idea indeed.
Seems nice, as I understand you connect and communicate with the server in `restapicamera.py` right? Seems like a very good solution! and only involves the interface layer itself.

beniroquai, 2021-11-13:
Yes, so there will be a server and a the client. I was not yet able to test
it since the OpenCV on the Pi made problems. But in theory this should work
ok with lower framerates and pixeldims. If this works, this really opens up
new possibilities for many of our setups since We could really control many
of our setups using ImSwtich. I just need to find somebody on our end to do
the programming. I'm lacking the skills here ;-)

>

kasasxav, 2021-11-14:
Cool! I've also been wanting to have our Nidaq not be so hardcoded, like in the lasers config where we set the digital lines, but rather one low level manager more, that could be either other device type.

closed 2021-12-07 by kasasxav