ImSwitch/ImSwitch #97: Behaviour First install (Data/Docs/Setups)

beniroquai, 2021-11-17:
Hey, Mr. Complain strikes again. I was wondering if you can help me out again - most likely for clarification only. 
I installed Imswitch on several machines now and the procedure how the `json` files are copied is always a bit confusing. This is most likely due to the issue that I want to have them inside the python package (changed the home directory). But as you already mentioned, this created some recursion with the folders and that's very messy. What do you think is the best practice to keep the `json` files (e.g. those for UC2) individual, but distribute them through Github too? 

Thanks a lot! :) 

PS: Congrats for your CZI funding! Worth it!

kasasxav, 2021-11-17:
Hi @beniroquai, I was wondering you can maybe try to name the folder something else than ImSwitch, maybe the recursion problem is related to the fact that the names are the same? But I actually don't know.

I think the best way would be to have it [here](https://github.com/kasasxav/ImSwitch/tree/master/imswitch/_data/user_defaults/imcontrol_setups), but then unless you have mockers for the devices there's a GitHub test that will fail. We can also try to do it in the docs somehow, either readthedocs or in GitHub (discussions for example).

I am thinking a lot about how to do this properly, as you said on Twitter, it would be nice to standardize the device layer of the different control software packages. I was also thinking if it would make sense to delegate the lower level classes you did to another project, some kind of library of openuc2 that you can then import in ImSwitch. I guess the managers would still have to be in ImSwitch but maybe everything else could be in another repo? I'm just thinking what's the best way in terms of sustainability. Let me know what's your view on this.

beniroquai, 2021-11-23:
Hey @kasasxav thanks for your thoughts! Side-loading the `JSON` files sounds reasonable. Could be a second Git with configurations, where different branches hold different setups? Not sure. 

Regarding low-level stuff. Yes. Totally. I was discussing that a lot with Richard Bowman from the Openflexure Project. They are very busy with putting together the Web-of-things standard for Labs ("LabThings", found [here](https://github.com/labthings/python-labthings). It offers a nice OpenAPI or Thingdescription as a Json that is readable by humans and a REST API. With this you can interact with this easily from any programming language. The Swagger API Converter/Generator can automatically generate the clients which is nice. 
Joining this with our ESP32 libraries ([UC2-REST](https://github.com/openUC2/UC2-REST)) would definitely be long-term goal/dream. However, for this I would probably need some people helping me. I'm not really a programmer ;-) 

What do think?

kasasxav, 2021-11-23:
That sounds good, I read the paper and I think it's super interesting, and definitely having an external library would be nice. For now I think having your fork is ok while we find a solution to integrate it, I think if you have your own library that would be more sustainable in case you want to use it with other software, for example, and maintain it.

I would say try to think if there is a way to make your devices common in some way, so in the future we can aim to have a manager like UC2DeviceManager that can control any type of device of your type and imports your library that handles the way it communicates, or something like that. I will think about this, I'm not sure what's the best way.