ImSwitch/ImSwitch #98: Saving recordings as TIFF

jacopoabramo, 2021-11-19:
I'm currently trying to get some recordings from my Ximea camera and save them as TIFF files. As far as I've understood only the snap feature allows the user to save images both on disk and viewer with TIFF formats. Is there any chance to save recordings as well?

beniroquai, 2021-11-22:
Hey, I'm facing the same issue. Also, when trying to safe images with the Allied Vision or Daheng Imaging cameras, I'm getting an error
```py
2021-11-22 10:13:44 ERROR [ExceptionHandler] Traceback (most recent call last):
  File "c:\users\uc2\documents\imswitch\imswitch\imcontrol\model\managers\RecordingManager.py", line 166, in run
    self._record()
  File "c:\users\uc2\documents\imswitch\imswitch\imcontrol\model\managers\RecordingManager.py", line 237, in _record
    dataset[it:recFrames, :, :] = newFrames[0:recFrames - it]
  File "h5py\_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
  File "h5py\_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
  File "C:\Users\UC2\Anaconda3\envs\imswitch\lib\site-packages\h5py\_hl\dataset.py", line 939, in __setitem__
    mspace = h5s.create_simple(selection.expand_shape(mshape))
  File "C:\Users\UC2\Anaconda3\envs\imswitch\lib\site-packages\h5py\_hl\selections.py", line 264, in expand_shape
    raise TypeError("Can't broadcast %s -> %s" % (source_shape, self.array_shape))  # array shape
TypeError: Can't broadcast (2, 1920) -> (2, 1200, 1200)
```

Surprisingly, I cannot debug that in Visual Studio Code. Not sure why the error is thrown, but the breakpoint is skipped.

Obviously a dimension mismatch is causing the problem.Most likely in get_chunk. Any idea how to solve this @kasasxav? 

Thanks a lot! :-)

beniroquai, 2021-11-22:
Maybe you can already help me out quickly. The function 

```py
    def getChunk(self):
        frames = []
        try:
            while True:
                im = np.array(self._camera.getLastChunk())
                frames.append(im)
        except Exception:
            pass
        return frames
```

has a while-loop that never ends (?). How come the continuous frame acquisition knows when to stop/break the loop? 

`self._camera.getLastChunk()` returns a single "RAW" frame.

With this I was able to to get rid of the above error. But I aggree with @jacopoabramo , that having a multipage TIF output would be nice too.

kasasxav, 2021-11-22:
Hi! 

I will work on it, I think it's definitely possible to save recordings also in TIFF and having them as a napari layer. We didn't need that feature so we only considered snaps, but now that you suggested it makes total sense.

And @beniroquai, the getchunk gets all the available frames at the moment of execution, so in the end it stops when there is no more frames (exception and pass). 

To fix your issue try to take a look at the dimensionality of newFrames, it should be a 3D array where the first index is the number of frames, it seems that you instead have a 2 dimensional matrix where X Y are in a single index as an array.

jacopoabramo, 2021-11-22:
Hey @beniroquai , I'm not sure that the `getChunk` function should be implemented in this way though. As far as I've understood, this function sort of grabs the amount of frames stored in a buffer, but this highly depends on the type of camera that you are using. I believe that this method here works fine with Hamamatsu cameras because users can directly read the overall frame buffer from the hardware FIFO and process them, but with other cameras it may not work as well. If you put an infinite loop in getChunk, ImSwitch should directly freeze because there's another loop hidden in the RecordingManager which calls it frequently to grab the latest acquired chunk of images (or in your case, just a single image). You should probably call `getLastChunk` only once and cast it to a `np.array` using the `np.stack` function, or at any rate something that can convert your frame to a (Z, X, Y) format. But that depends on how the getLastChunk function spits out its data.

beniroquai, 2021-11-22:
Hey both, thanks for your support! :) 
I must admit that I copied the function from the `XimeaManager`. The `np.array(self._camera.getLastChunk())` function really just outputs a single frame. So I'm still not 100% sure when the `while()`-loop will break since there will be frames available all the time. Not sure if I can address the buffer from the Daheng/Allied Vision cameras to be honest. (At least not with my programming skills ;-) )

jacopoabramo, 2021-11-22:
> Hey both, thanks for your support! :) I must admit that I copied the function from the `XimeaManager`. The `np.array(self._camera.getLastChunk())` function really just outputs a single frame. So I'm [...]

Hey @beniroquai , you make me blush if you use my work, I feel useful now!

I would suggest then to just call something like this:

```python
np.stack([np.array(self._camera.getLastChunk(), (sizeX, sizeY)])
```

In this way you grab the latest frame from the camera, convert the raw data into your sizeX and sizeY format (Numpy should be handling that with no problem), wrap it into a list and call np.stack which will create the (Z, X, Y) stack that `getChunk` needs to return.

Or you could also call `np.array(self._camera.getLastChunk(), (1, sizeX, sizeY))`, and I guess it should be the same...? Try it out maybe.

beniroquai, 2021-11-22:
Whoop whoop

[image]

Right in time for doing some SMLM with UC2 :D 

Thanks a lot!

kasasxav, 2021-11-22:
That looks great! Thanks a lot for the patience and @jacopoabramo for helping out :)

I agree though with @jacopoabramo that it might not be the best way to rely on the buffer of the camera itself, but I thought it would be more resilient since it will be implemented in a lower level language, but it's a big assumption. Do you have suggestions in how to implement it better?

jacopoabramo, 2021-11-22:
> Do you have suggestions in how to implement it better?

@kasasxav in my opinion there are a couple of solutions which we could take into account for not only the recording but also the live view.

If you recall [here](https://github.com/kasasxav/ImSwitch/discussions/91), I suggested the integration of a common local RAM buffer (possibly a circular buffer - I should have probably followed up on this, I apologize!). Now this here shouldn't be forced to use upon the user, because as you noted the Hamamatsu camera can easily get all the required frames from it's local buffer, but it could be nifty for cameras which don't have this option. Again this is only in reference to the fact that up until now I didn't find any way to recover the frames from the Ximea as you do in the Hamamatsu.

SO! That being said, option one:

- Each camera manager could have instantiated a local frame buffer, and it's size could and should be controlled by the settings widget (I believe that something similar can be done also in Micro-Manager); everytime a new frame is acquired during live view it's stored in this buffer, and then the `getLatestFrame` simply pops from this buffer. This raises a couple of questions, especially in the risk of losing frames because the live view is not refreshed fast enough.
  - Ideally we should have a recording thread which stores the frames in the buffer, and then these are popped from another thread which hosts the RecordingManager, but I do believe that this is already taken care of not only for the recording but also the live view.
  - Due to this, the frame buffer should be thread-safe, and python already provides such containers in the `collections` package.
- In case a recording is requested, the frame buffer should be cleared and filled only with the frames from the requested recording. Or, eventually, a secondary buffer could be added only for recording (but it may become quite expensive in terms of memory usage).

Option two:

- We discard the frame buffer implementation and we provide the `getChunk` function as a context manager function, which yields the acquired frames one by one. I don't honestly know how much this differs from the frame buffer idea but it's probably more viable and easy to implement.

I can't stress enough that this is only due to my experience with the Ximea camera: we have a reeeeally fast camera that can reach up to 10k FPS with small ROIs, and the biggest problem is to make sure that while in free running no frames are lost. This is actually how I handle the live and recording acquisitions in my napari-live-recording plugin, so you could also check it out if you want.

I don't know if this comment makes you understand what I'm trying to say, so feel free to point out any flaws in this in your opinion.

jacopoabramo, 2021-11-22:
Oh, by the way @kasasxav , I did have some issues in saving the recordings as TIFF stacks in napari-live-recording: whenever the size became bigger than 4 GB, the stack should be written in BigTIFF format. I think that this can be done using the `tifffile.TiffWriter` class in the `tifffile` package, but this is still an open issue on my plugin - see [here](https://github.com/jacopoabramo/napari-live-recording/issues/7)

beniroquai, 2021-11-22:
Fresh from the UC2 STORM setup (Yes! It blinks!). Next step: Make it work yet a second time
https://user-images.githubusercontent.com/4345528/142939754-ba405fe2-4370-463a-b36f-29b0e3619b69.mov
Thanks a lot for your help! :-)

kasasxav, 2022-02-03:
Hi @jacopoabramo, I will take into account your comments and hopefully integrate this some day, but since it's not a priority I will close the issue for now!

closed 2022-02-03 by kasasxav