ImSwitch/ImSwitch #99: Feature: Having keyboard/joystick control

beniroquai, 2021-11-22:
First of all: Today's experiment wen't really well. ImSwitch helped a lot. 
Now I was wondering, if we can integrate haptic control for moving the stage/focus. I'm more-less happy with my PS3 controler implementation for the cellSTORM project (purely ESP32+Bluetooth microcontroller system) and was wondering if there is any way to sneak in external control using e.g. arrow-keys or even the PS3 sixaxis controller. 
There is a simple way to include it in Python using `pygame` (e.g. [Tutorial](https://theraspberryblonde.wordpress.com/2016/06/29/ps3-joystick-control-with-pygame/)). We have "played" with it in a micromanipulator project [here](https://github.com/openUC2/openCM2_software/blob/master/src/controller2motor.py). 

What do you think? Is it worth having a look at it? Not sure where to start to be honest.

jacopoabramo, 2021-11-23:
Hi @beniroquai , just out of curiosity what board are you using for the ESP32? A custom board or an open-source project of some kind?

jacopoabramo, 2021-11-23:
As for your specific question, let me see if I got it right: you want to move your stages via your PS3 controller using a wireless communication (i.e. Bluetooth)? So you need a receiver which transmits the move commands to the stages?

kasasxav, 2021-11-23:
Hi :) Glad that the experiment went well
oh interesting... It's something I never thought about. I would say the PS3 sixaxis controller in the end is a device so it should have a Manager, and I think that there should be a thread inside to "listen" for events (like your while in the example).  As `managerProperties` you would need to link the different directions to different positioners, for example horizontal to `stageX`, vertical to `stageY` and the other axis to` stageZ`. 

How to make the axis manager control de positioner manager is by adding PositionersManager in initialization in the master controller I guess. Then when you move an axis you should do positionersManager.execOn(function, stage) meaning it will execute the function move in the stage that you want. 

I don't know if the way I wrote this is clear or a bit convoluted but let me know what you think.

kasasxav, 2021-11-23:
As for how to do this in a pragmatic way, I would:
1. Create a Manager for the joystick, that has a thread and detects when it moves in different directions and just logs it, so you can check if it's doing it right.
2. Add the Manager to ImSwitch and your config file, and test if it works. Maybe in this case it will be similar steps as @jacopoabramo did for the `PulseStreamerManager`, since it's a Manager totally different from the ones we had before. In `managerProperties` have like `horizontalDirection: name of Stage`, etc.
3. Then when initializing the Manager (I think in `MasterController`), add the `positionersManager` in the construction.
4. Then in the functions that you did in the manager call `positionersManager.execOn(func, nameOfStage)` like I said in my previous comment (fetching the `nameOfStage` from the config file, that the Manager already has,` setupInfo`).

kasasxav, 2021-11-23:
With the keyboard I think it's a bit different, because you could use the shortcut function (we have a decorator to create shortcuts), so you could have the arrow keys like that. Maybe easier?

jacopoabramo, 2021-11-23:
> With the keyboard I think it's a bit different, because you could use the shortcut function (we have a decorator to create shortcuts), so you could have the arrow keys like that. Maybe easier?

@kasasxav I don't know if it makes too much sense, but what about using slots connected to the key-pressed signal from pyQt to emulate this behaviour? How hard would it be to implement?

jacopoabramo, 2021-11-23:
Man am I dumb! Thanks that's really nifty!

kasasxav, 2021-11-23:
haha no it's fine, sorry I didn't explain it properly :P

jacopoabramo, 2021-11-23:
Honestly my mind was racing with something like this:

- the joystick can send data to the ESP32 which acts as a bluetooth receiver and serial transmitter to the setup;
- the serial could act as a sort of virtual keyboard and sending commands to go move the stage as you want;

As it's an ESP32, I think you could also use the Arduino libraries (I think they are compatible? But I'm not sure) and do something like [this](https://www.arduino.cc/reference/en/language/functions/usb/keyboard/keyboardpress/). So you would catch two birds with one stone: you can have both the actual keyboard to move the stages as you want, OR use the joystick.

But I don't have enough details on how the PS3 sixaxis controller works, so maybe I'm way off on what @beniroquai wants to do.

Also sorry about interjecting in the topic, but I'm invested in learning more about the ESP32 for a personal side project so I kinda wayed myself in y_y°

beniroquai, 2021-11-24:
Hey, sorry, we were busy doing some measurements in the lab. Had to quickly adapt a new camera. Worked like a charm! :) 

I'll work through the messages:

> Hi @beniroquai , just out of curiosity what board are you using for the ESP32? A custom board or an open-source project of some kind?
The ESP32 is similar to an Arduino, costs 5€ and interfaces to its environment (mostly) through WiFi or Bluetooth (i.e. IoT). We have a bunch of them and you can do cool stuff with it. It has two cores, hence you can do Galvo control while serving a RESTFUL API. 

> As for your specific question, let me see if I got it right: you want to move your stages via your PS3 controller using a wireless communication (i.e. Bluetooth)? So you need a receiver which transmits the move commands to the stages?
Yes, PS3 controllers use Bluetooth. That's how we interface it with the ESP's at the moment. 

> Hi :) Glad that the experiment went well oh interesting... It's something I never thought about. I would say the PS3 sixaxis controller in the end is a device so it should have a Manager, and I [...]
Yes, I tried it in a very dirty way using `ImScript` and PyGame. It worked, but it felt like blocking the main thread? Definitely not the best solution.. But cool actually ;) 

> As for how to do this in a pragmatic way, I would:
> 
> 1. Create a Manager for the joystick, that has a thread and detects when it moves in different directions and just logs it, so you can check if it's doing it right.
> 2. Add the Manager to ImSwitch and your config file, and test if it works. Maybe in this case it will be similar steps as @jacopoabramo did for the `PulseStreamerManager`, since it's a Manager totally different from the ones we had before. In `managerProperties` have like `horizontalDirection: name of Stage`, etc.
> 3. Then when initializing the Manager (I think in `MasterController`), add the `positionersManager` in the construction.
> 4. Then in the functions that you did in the manager call `positionersManager.execOn(func, nameOfStage)` like I said in my previous comment (fetching the `nameOfStage` from the config file, that the Manager already has,` setupInfo`).
> 
> edit: this is a suggestion but there could be other ways to do it, let me know if you have other ideas.

Sounds perfect. And I just found out that @jacopoabramo works in the same institute as I do :D .. small world. Let'S meet! ^^

> yes that's how the shortcuts are done, so basically you do, for example:
> 
> ```
> @shortcut('Ctrl+L', "Liveview")
>     def toggleLiveviewButton(self):
>         self.liveviewButton.toggle()
> ```
> 
> And then when you press Ctrl+L that function will be executed, and automatically included in the menu.

Where would I implement that? I'm sure you have that in the code somewhere already, so I'll read further ;) 

> @shortcut(

Let's chat. Roomnumber 143 ^^

kasasxav, 2021-11-25:
Nice! And good that it worked in the ImScript.
Yeah the thing I suggested it was more like treating it as a device, but I liked the alternative of @jacopoabramo to treat it as a keyboard.
So cool that you are in the same institute! haha
Regarding the shortcut, in theory if you add that like in any function it will work no matter where the function is, if I remember correctly, but better do it in the stuff that is in the `WidgetController`.

jacopoabramo, 2021-11-25:
@beniroquai yes I noticed that too just yesterday, it's awesome! I'll write you an e-mail v.v I'm in another building at the moment for *reasons*, I'll explain it to you :p

beniroquai, 2021-11-30:
Hey @kasasxav, I started implementing the shortcut-thing [here](https://github.com/beniroquai/ImSwitch/blob/master/imswitch/imcontrol/model/managers/positioners/GRBLStageManager.py#L58). I'm attempting to add the following code:

```py
    # test arrow key input
    from PyQt5 import QtCore
    @shortcut(QtCore.Qt.Key_Up, "Move up")
    def toggleLiveviewButton(self):
        self.move(value=100, axis = "X")
    @shortcut(QtCore.Qt.Key_Down, "Move down")
    def toggleLiveviewButton(self):
        self.move(value=-100, axis = "X")
    @shortcut(QtCore.Qt.Key_Left, "Move left")
    def toggleLiveviewButton(self):
        self.move(value=-100, axis = "Y")
    @shortcut(QtCore.Qt.Key_Right, "Move right")
    def toggleLiveviewButton(self):
        self.move(value=100, axis = "Y")                        
   ```

Is this all? 

Before I start implementing what you suggested here:

> As for how to do this in a pragmatic way, I would:
>
> 1. Create a Manager for the joystick, that has a thread and detects when it moves in different directions and just logs it, so you can check if [...]

May I ask you if you have some super basic code snippet I could use as a reference? That would be very helpful! :-) 

Thanks a lot!

kasasxav, 2021-11-30:
Hey @beniroquai, I answer the second part later.
Regarding the first yes, I would though prefer to have it [here](https://github.com/beniroquai/ImSwitch/blob/master/imswitch/imcontrol/controller/controllers/PositionerController.py) because it's a View thing kind of (since Qt is involved also). For example, if you have 2 objects of that stage all of them will move.
Then, instead of `self.move(value, axis)` you need to do `self.move(positionerName, axis, value)`, so from the positioners you have, you should set which one you want to move.

kasasxav, 2021-11-30:
Hi again,

Now I have a bit more time, so what I meant was to do it in the class `PositionerController` and including it in the config file. If you want to follow these steps then you can do a small pull request to the main (if possible separated than the one you did and with only this change :P):

1.  In [SetupInfo.py](https://github.com/kasasxav/ImSwitch/blob/master/imswitch/imcontrol/model/SetupInfo.py), line 80 add:

              `forShortcut: Optional[Union[List[str], int]] = None`
              
 this means that in your config file, when adding each positioner you can _optionally_ add, for example:
              `forShortcut: {"X", "Y"}` in one, and `fortShortcut:{"Z"}` in another, or I assume in your case you have all of them in one.

2. In [PositionerController](https://github.com/kasasxav/ImSwitch/blob/master/imswitch/imcontrol/controller/controllers/PositionerController.py), in the init (line 23, inside the for loop):

```
              if pManager.forShortcut is not None:
                for axis in pManager.forShortcut:
                  self.shortcutPositioner{pManager.forShortcut[axis]} = pName
```

3. Then you can make your functions in `PositionerController`
```
@shortcut(QtCore.Qt.Key_Up, "Move up")
    def key_moveXup(self):
        self.move(self.shortcutPositioner{"X"}, "X", 100)
    @shortcut(QtCore.Qt.Key_Down, "Move down")
    def key_moveXdown(self):
        self.move(self.shortcutPositioner{"X"}, "X", -100)
    @shortcut(QtCore.Qt.Key_Left, "Move left")
    def key_moveYleft(self):
        self.move(self.shortcutPositioner{"Y"}, "Y", -100)
    @shortcut(QtCore.Qt.Key_Right, "Move right")
    def key_moveYright(self):
        self.move(self.shortcutPositioner{"Y"}, "Y", 100)                     
    @shortcut("-", "Move Z up")
    def key_moveZup(self):
        self.move(self.shortcutPositioner{"Z"}, "Z", 100)                  
    @shortcut("+", "Move Z down")
    def key_moveZdown(self):
        self.move(self.shortcutPositioner{"Z"}, "Z", -100)      
```

The other suggestion, if you can do so that when you press the joystick you can detect it as a QtCore.Qt.Key something, that would be amazing, otherwise I think I don't have any code stuff, but maybe check the commits of Jacopo in the commit history of this repo in order to understand what are the steps better.

jacopoabramo, 2021-11-30:
I could maybe illustrate what I did for the PulseStreamer on our call on friday @beniroquai , hoping that I remember what I did @.@

kasasxav, 2021-12-07:
I close this for now but open new issues if you find problems or issues while you implement this :)

closed 2021-12-07 by kasasxav