ImSwitch/ImSwitch #100: Question: Limit in ROI?

beniroquai, 2021-11-30:
Hey, me again. I just implemented the ROI functoin in the Daheng camera and I cannot set any valuehigher than 256 or 100 for width/height in the ImSwitch ROI Manager. Is this intended? I could not find any code that would descrbe that. Perhaps I missed some setting in the Json?

kasasxav, 2021-11-30:
Isn't it the limit you have in your `setROI` that it does max between 256 and the number you input?

kasasxav, 2021-11-30:
gxipycamera.py  line 150

kasasxav, 2021-11-30:
In the hamamatsu we can put up to 2048 (fullChip), so I think it will be somewhere in your manager. It could also be the` fullShape `parameter being too small.

beniroquai, 2021-11-30:
Damn! You are so smart! 
Indeed, I have read the sensor width/height from the cropped camera sensor. After each restart, it's a different value and limiting the ROI. Perfect! Thanks :)

closed 2021-11-30 by beniroquai