ImSwitch/ImSwitch #102: [IDEA] GUI for mapping e.g. well plate coordinates

beniroquai, 2021-12-07:
Hey, 
I'm starting to love the software a lot and after our last meeting with @jacopoabramo, I'm keen on implementing the keyboard stuff with a dedicated Manager. 
Something I was thinking of, since a collaborator would like to have a system for well plate scanning, would be a map, where you click on a well location (e.g. X/Y 3,4) and the microscope automatically moves there. Did you ever encounter such a "feature" as helpful? Would this be implemented as a completely new "tab" similar to the `imcontrol` for python interaction or how would that work? 

Thanks! :)

kasasxav, 2021-12-07:
Hi @beniroquai ,
nice that you are finding the software useful! I'm currently thinking of some architectures ideas so that your contribution can be integrated :) I will ask you at some point for a meeting if it's ok.

I think that looks good, the map would be in the GUI right? I would create a new Widget actually inside `imcontrol`.  I think it's the only part of ImSwitch you have not touched yet :P Maybe you can check one of the other widgets to know how it would be done? I suggest the [PositionerController](https://github.com/kasasxav/ImSwitch/blob/master/imswitch/imcontrol/controller/controllers/PositionerController.py) and [PositionerWidget](https://github.com/kasasxav/ImSwitch/blob/master/imswitch/imcontrol/view/widgets/PositionerWidget.py)

But in general, you would need to create `WellPlateWidget` and `WellPlageController `(or other names) which should be subclasses of `Widget` and `WidgetController`. Then, in `availableWidgets` in the config file you should add `WellPlate`, and then it should be integrated. I have to say I'm also looking into improving the way people can do their own widgets since it might not be super-intuitive, but for how it is now I would suggest the following steps:

1.  Code the `WellPlateWidget`, only with the buttons and GUI components, and then connect the QT signals to external signals like it's done in other widgets. For example: `self._someButton.connect(self.sigSomeButtonClicked)`.
2. Then you need to do the `WellPlateController`, which because it being a subclass of `WidgetController` will have the `WellPlateWidget` as `self._widget` and the master controller as `self._master`. You need to connect those signals with functions in this class, and then here you write everything that is more "controll stuff".
3. Think of the managers you need to implement, probably you already have them as `Positioners`. You can access them through `self._master.positionersManager[positionerName].move()` for example. Here you would do essentially like in the `PositionerController`.

Let me know if you have questions or run into problems!

kasasxav, 2021-12-07:
You also need to add its name in the [main view](https://github.com/kasasxav/ImSwitch/blob/master/imswitch/imcontrol/view/ImConMainView.py) either in the right widgets or left.

beniroquai, 2021-12-08:
Great! I think with this info I can make it work. 
I found [this](https://github.com/Kabilan108/WellPlate) code for the Platelayout and will use it for now. 

I think in the long run, it would be great to have something like an installable plugin (?) for managers so that you don't need to mess with the code. Perhaps the same with cameras etc. But this sounds crazy to me. The code is modular enough to do it, but I think making it foolproof (like for those people who cannot code like me), is another thing. 

Will keep you updated! :)

beniroquai, 2021-12-08:
And yes, let's meet at some point! :)

beniroquai, 2021-12-08:
So far I had to add:
`from .WellPlateController import WellPlateController`
 in 
`.\ImSwitch\imswitch\imcontrol\controller\controllers\__init__.py

and 
`from .WellPlateWidget import WellPlateWidget`
in 
`imswitch\imcontrol\view\widgets\__init__.py`

correct?

beniroquai, 2021-12-08:
Well..
[image]

kasasxav, 2021-12-08:
Oh great! That was quick haha You are like the best reviewer ever, tested everything. Thank you for trying 🤗 Let me know if you find difficulties

beniroquai, 2021-12-08:
As said, I have no clue what I'm doing, but it somehow works.
In case the GUI gets cluttered, is there a chance to work with "Tabs"?

>

jacopoabramo, 2021-12-08:
Something like this could also be useful for a sampler mover. I encountered some months ago a request from a company to integrate a GUI which describes a GRID of samples of different sizes and the user would be able to control the current selected sample under analysis.

closed 2022-02-03 by kasasxav