ImSwitch/ImSwitch #103: Question: Are hardware actions blocking the main / GUI thread?

beniroquai, 2021-12-17:
Hey @kasasxav, 
today I encountered an interesting problem (perhaps not a problem?). 
It all works nice with the setups we have and I'm really happy, so my plan is now of course to get some more nice data out of it ;-) 
I wrote a small script to take a z-stack in red and blue:

```py
import numpy as np
import time

def snap_zstack(N_images=50, dz=4):
	axis = "Z"
	api.imcontrol.movePositioner(positioner[0], axis, -(N_images*dz)//2)

	for i_pos in range(N_images):
		api.imcontrol.movePositioner(positioner[0], axis, dz)
		#time.sleep(.3)
		api.imcontrol.snapImage() 
	time.sleep(.5)
	api.imcontrol.movePositioner(positioner[0], axis, -(N_images*dz)//2)

	
mainWindow.setCurrentModule('imcontrol')

laserName = api.imcontrol.getLaserNames()
positioner = api.imcontrol.getPositionerNames()

laser_488 = '488 Laser'
laser_635 = '635 Laser' 

# snap GFP
api.imcontrol.setLaserActive(laser_635, False)
time.sleep(.3)
api.imcontrol.setLaserActive(laser_488, True)
time.sleep(.3)

snap_zstack()
#api.imcontrol.snapImage() 

# snap AF647
api.imcontrol.setLaserActive(laser_488, False)
time.sleep(.3)
api.imcontrol.setLaserActive(laser_635, True)
time.sleep(.3)

#api.imcontrol.snapImage() 
snap_zstack()

# turn off lasers
api.imcontrol.setLaserActive(laser_488, False)
api.imcontrol.setLaserActive(laser_635, False)
```

I changed the way I talk to the ESP's and faced a - perhaps - more general problem. I wanted to make sure that everytime I request a position with the Z-stage, the positioncontroller waits until it reaches this position (vs. having a thread that just send the request and does not wait). However, it seems, when I'm waiting for the motor to move, the main thread is blocking, hence no GUI actions are possible. Is this the correct behaviour ? Perhaps it's also an issue on my end. Not sure how to deal with it. When going from red->blue laser, the filter changes, so that takes a few seconds. The GUI becomes inresponsible. But putting the hardware in a thread won't change this I think, or? I have to wait until the hardware reaches its desired location..

Best
Bene

kasasxav, 2021-12-18:
Hi @beniroquai, I think since you have a time.sleep() that might be blocking the main thread right? This is something I haven't thought about much but when we for example do a scan, we have signals like `api.imcontrol.signals().scanEnded` for example and then we do this: https://github.com/kasasxav/ImSwitch/blob/cb65c784c641f920f3060ed691c9b6292b135f1f/imswitch/_data/user_defaults/scripts/example_record.py 

Can you do something similar? Do you know from the stage when it is set to the position?

beniroquai, 2021-12-18:
Ok. I understand. But I think the main problem I see is that, if you have a command send through e.g. USB, REST or whatever and you wait for the hardware to arrive at whatever state, this check in the HardwareManager makes the GUI stuck. It freezes. Seems like, there is no background thread running for this or? I bet I'm wrong, but that's how it feels now ;-) 
I'll double check!

kasasxav, 2021-12-19:
oh really? But is the manager really waiting for the hardware to arrive at a certain state? Do you have a specific example of this? I thought that the manager would just send the command to the hardware and not wait, not sure. Maybe we can take it in our meeting and in case this is happening could find some solution doing more threads

beniroquai, 2021-12-22:
Hey, I'm really bad in describing things. So I better record a video ;-) 

Here you go:
[[image]

So while changing the filter (which I do when turning on the filter), the process is: 
1. HTTP request is send to the ESP32 that tells it: turn off laser, change filter, turn on laser
2. Once done, it sends a response (json) á la REST API and tells ImSwitch: "I'm done"

It seems the main thread is blocking. But this should be independent from any hardware, or?

kasasxav, 2021-12-22:
Hi @beniroquai, thank you!! You explained it well sorry, just trying to follow up with your super cool work :P
Could you tell me where 2. is happening in the code? I think you're right, probably the waiting until it's done it's blocking the main thread, so we could think of doing it in a separate thread.

kasasxav, 2021-12-22:
I'm guessing somewhere in LabThingManager but I'm not sure of the syntax, but I think maybe there one could work with threads?

closed 2022-02-03 by kasasxav