ImSwitch/ImSwitch #104: Executing "external" scripts without imscript?

beniroquai, 2022-01-18:
Yet another question/idea. 
I managed to get ImSwitch incl. Napari working on the Jetson. Wuhu! But: without the imscript and QSci plugin. I was wondering if there is any chance to provide scripts with an external editor and execute them by hitting a button in imswitch? Building Qsci and installing in an environment seems a bit complicated for the vast majority of people (including me) ..  ;)

kasasxav, 2022-01-18:
Hi @beniroquai,
I'm thinking this could be possible. But if QSci is the problem, that is only used for the script editor [EditorView](https://github.com/kasasxav/ImSwitch/blob/master/imswitch/imscripting/view/EditorView.py), [EditorController](https://github.com/kasasxav/ImSwitch/blob/master/imswitch/imscripting/controller/EditorController.py) in `imscripting`. But there is also a console there, so I'm thinking if it would make sense to try to temporary remove the editor (I can create a branch so you try) and you can execute scripts from the console, for example:

`script = importScript(path)
script.run()`

You would need to have functions inside the script that you can call, maybe just `main` or so. 
Do you think this could be a good solution? In that case, we could later on make it optional to have the editor, and we could also include just a button for running a script but without QSci.

kasasxav, 2022-01-18:
[image]
Like this, there you also have access to the` api `and everything. 
branch: [modified-imscripting](https://github.com/kasasxav/ImSwitch/tree/modified-imscripting)

kasasxav, 2022-01-18:
One could modify that even more, adding another editor that might be easier to install (I'm not so familiar with editors), or having some other sort of GUI, you can add your own widget if you want to have a specific way of dealing with scripts.

beniroquai, 2022-01-18:
You are a genius! Will try that when I'm back home. 
It's working nicely with the jetson though. Fun! ;-)
And: Well done!

closed 2022-02-03 by kasasxav