ImSwitch/ImSwitch #108: Cobolt0601 bypasses safety key

jacopoabramo, 2022-01-28:
Greetings,

I was debugging the reason why using the python-microscope layer to control my [Cobolt 06-MLD (laser)](https://hubner-photonics.com/products/lasers/diode-lasers/06-01-series/) was not working (see [#234](https://github.com/python-microscope/microscope/issues/234)), while instead the Cobolt0601LaserManager was apparently working well. Curiously enough, the safety key for the Cobolt laser was turned off. But I was still able to control the laser via ImSwitch. I don't know if this is an issue on it's own but I thought it was worth to mention.

kasasxav, 2022-01-28:
Oh I had no idea about that ! Thanks for mentioning, I will test it with our lasers. But since you did the PM integration I will also try to use that and maybe we can replace the managers :)

jacopoabramo, 2022-01-28:
One thing I should mention too (which I just remembered) is that the laser may need a timeout in order to avoid the same behaviour mentioned in the issue. I'll fix this somehow but just for your safety on your setup if you can patch the manager for a quick test you can see if it behaves as I described.

kasasxav, 2022-03-02:
Hi, sorry for the delay.
Yes I confirm that it seems to bypass the safety key, because of the lantz driver somehow.
I am trying though to run it using the python microscope manager and I ran with an error, I was wondering if you have had it before?

`  File "D:\Github\imswitch\imswitch\imcontrol\model\managers\lasers\PyMicroscopeLaserManager.py", line 26, in __init__
    self.__laser = getattr(laserClass, driver[1])(self.__port)
  File "D:\anaconda3\envs\imswitch\lib\site-packages\microscope\lights\cobolt.py", line 65, in __init__
    self._max_power_mw = float(self.send(b"gmlp?"))
ValueError: could not convert string to float: b'OK'`

jacopoabramo, 2022-03-02:
> Hi, sorry for the delay. Yes I confirm that it seems to bypass the safety key, because of the lantz driver somehow. I am trying though to run it using the python microscope manager and I ran with an [...]

Yes, it was already mentioned in this [issue](https://github.com/python-microscope/microscope/issues/234). Apparently the Cobolt driver requires a timeout which with our lasers is too small. They already increased the default timeout for the serial communication but I don't know when they are going to release the latest changes. Best thing you can do is to patch microscope on your end and increase the timeout. Otherwise a good thing would be to pass arguments to the constructor of the laser from the configuration file, trying to keep a generalized approach.

kasasxav, 2022-03-02:
Ah right! Thanks, just realized it's the same issue you mentioned before.

Alright then, thank you. I think I don't mind waiting for the release and when it's done I will update to use your manager instead of Cobolt0601.

I close for now then :)