ImSwitch/ImSwitch #110: Possibility to scale the whole GUI?

beniroquai, 2022-02-04:
Do you know if there is a QT preference that allows to scale the entire GUI by e.g. 50%? On my MAC and on the Jetson only a small subset is visible. Must be a DPI-related setting, or? 

[image]

beniroquai, 2022-02-22:
Ping! ;-)

kasasxav, 2022-02-23:
Hi @beniroquai, I checked but it's a bit tricky. The widgets themselves should already be scalable, if you click in between the image and the docks in the right, for example, you can drag it to the right or even hide them. 

But I think the issue is that upon start of ImSwitch, how it scales the left, center and right widgets doesn't look good when the resolution of the screen is low. I don't really know how to fix it, but could you maybe try changing the stretch in the [main view ](https://github.com/kasasxav/ImSwitch/blob/master/imswitch/imcontrol/view/ImConMainView.py)(line 109-115) to give more space to the image? 

Do you have other suggestions?

beniroquai, 2022-02-23:
Hey thanks for commenting! 
On Windows there is an easy fix. You just need to scale the GUI appearance  in the windows settings (e.g. 150% -> 100%), but in Linux/Mac that's rather difficult. I was playing with some global QT5 configs before starting the GUI [here](https://github.com/beniroquai/ImSwitch/blob/master/imswitch/imcommon/applaunch.py#L27) but it does not seem to have any effect. It'S an OS thing I suppose, but I could not yet find any documentation for this problem. Usually it'S the other way around. GUI elements too small..

beniroquai, 2022-03-24:
For the documentation..
In GNOME you can do this:
```
UC:~/ImSwitchConfig$ xrandr --listmonitors
Monitors: 1
 0: +*eDP-1 1920/344x1080/193+0+0  eDP-1
UC:~/ImSwitchConfig$ xrandr --output eDP-1 --scale 1.2x1.2
``` 
this scales the whole GUI. Same for Windows, but in Window Settings.

kasasxav, 2022-03-24:
Thank you! Sorry that I didn't find a better way for this.

kasasxav, 2022-04-29:
Not really, sorry :l I close from now, if I find a better solution I will let you know!

closed 2022-04-29 by kasasxav