ImSwitch/ImSwitch #112: Exporting Metadata in a unified format?

beniroquai, 2022-02-22:
Hey, I just listened to a talk by Caterina Strambia de Castilla about QUEREP. Do you think it's a wise decision to eventually think about adapting their  format into the metadata format?
Some more information can be found [here](https://wu-bimac.github.io/MicroMetaApp.github.io/)

kasasxav, 2022-02-23:
Hey, I think it's a great idea. I have thought about it before, now we are using hdf5 files and saving all the metadata from the widgets (laser powers, stage positions, scan values, etc), but it would be nice to upgrade it to what is done currently in the literature. I will keep it in mind, I don't know if I will have time though to do something soon though :l

beniroquai, 2022-02-23:
Perhaps we can get in contact with these people and think about a unified python module that takes a python dictionary and converts it into their "standard". Seems like they are very much up for that. Should I ask her?

kasasxav, 2022-02-23:
Sounds good :) I think it's a very good innitiative indeed!

strambc, 2022-03-01:
It would be great to work together on this. In order to do so.

As part of QUAREP-LiMi WG7 and in particular, as part of Sub-Group 3 on Metadata Capture and Storage, we are planning multiple tools but we do not have any concrete yet. 

It would be great if you could join to bring your needs and potential solutions you propose. 

This is the link to find out what we are up to: https://quarep.org/working-groups/wg-7-metadata/

This is the link to join QUAREP-LiMi: https://quarep.org/members/become-a-member/