ImSwitch/ImSwitch #113: Getting return value from Pyro5 server?

beniroquai, 2022-02-26:
Hey @kasasxav, finally I found some time to "remote" control the Pyro server. 
I was wondering if it is somehow possible to retrieve a return value from one of the signals. In fact, I was thinking to retrieve a "snap" as a Numpy array. For this I wrote a new function called `snapNumy` that just returns an image. 
However, it seems it's not trivial to return the frame through the underlying QT5 structure which I think is implemented [here](https://github.com/beniroquai/ImSwitch/blob/master/imswitch/imcontrol/controller/ImSwitchServer.py#L17). Do you know of any good alternative to realize this?

kasasxav, 2022-03-04:
Hey, sorry it's a bit work in progress :P Will try to be done with it soon!
For returning images in ImSwitch itself is quite easy, I have done it, but I'm working on defining how Pyro5 will serialize np arrays

kasasxav, 2022-03-04:
I have some problems when registering the serializers in the client side

beniroquai, 2022-03-04:
Ok, I was not able to get any return on the pyro side at all. Did you check different encoding (endian blabla? ;)

kasasxav, 2022-03-08:
Alright now I think it should work, in the pyro-functionality code. 
The idea is that you add in the config file the name, host, port you want optionally (otherwise it initializes to default values: PYRO:ImSwitchServer@127.0.0.1:54333). You can check in SetupInfo ([here](https://github.com/kasasxav/ImSwitch/blob/pyro-functionality/imswitch/imcontrol/model/SetupInfo.py#:~:text=Pulse%20Streamer%20hardware.%20%22%22%22-,%40dataclass(frozen%3DTrue),port%3A%20Optional%5Bint%5D%20%3D%2054333,-%40dataclass_json(undefined%3DUndefined) and [here](https://github.com/kasasxav/ImSwitch/blob/pyro-functionality/imswitch/imcontrol/model/SetupInfo.py#:~:text=Pulse%20Streamer%20settings.%20%22%22%22-,pyroServerInfo%3A,-PyroServerInfo%20%3D%20field(default_factory)) for the syntax.

From the client side you need to also do some stuff:

`import Pyro5.api
from _serialize import register_serializers
from useq import MDASequence

register_serializers()
uri = 'PYRO:ImSwitchServer@127.0.0.1:54333'
imswitchServer = Pyro5.api.Proxy(uri)
imswitchServer.exec("ViewController", "liveview", [True])
image = imswitchServer.get_image()`

Same _serialize as [here](https://github.com/kasasxav/ImSwitch/blob/pyro-functionality/imswitch/imcontrol/controller/server/_serialize.py). Check it out and let me know.

kasasxav, 2022-03-08:
How I did the `get_image` function is that you are supposed to first start the liveview and then you get the last image. You can do it differently if you want to just snap one image, but I think essentially you can do the same I did but change the way you retrieve the image from ImSwitch.

beniroquai, 2022-03-10:
Thanks a lot! I'll dive into that right now! I think it would be handy to have a small client that remote controls ImSwitch. Will work on that!

beniroquai, 2022-03-10:
First update: I'm testing the above "client code" and run into the first issue when using the `_serialize` function from this repo:

```py
>>> imswitchServer = Pyro5.api.Proxy(uri)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/anaconda3/envs/imswitch/lib/python3.8/site-packages/Pyro5/client.py", line 71, in __init__
    raise ValueError("unknown serializer configured")
ValueError: unknown serializer configured
>>> imswitchServer.exec("ViewController", "liveview", [True])
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'imswitchServer' is not defined
>>> image = imswitchServer.get_image()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'imswitchServer' is not defined
>>> 
```

Does this sound familiar? ;-)

kasasxav, 2022-03-10:
Yes sorry, was the main error I had that took me forever to fix haha
Remove this line from _serialize `Pyro5.config.SERIALIZER = "msgpack"`. It should only be in the server and not in the client :) I will change it in the file.

kasasxav, 2022-03-10:
Now it should be fine with #[891790f](https://github.com/kasasxav/ImSwitch/commit/891790fce9b0ffd929a30452852d231544fe5b92)

beniroquai, 2022-03-10:
Bähm! 
```py
>>> import Pyro5.api
>>> from _serialize import register_serializers
>>> 
>>> register_serializers()
>>> uri = 'PYRO:ImSwitchServer@127.0.0.1:54333'
>>> imswitchServer = Pyro5.api.Proxy(uri)
>>> imswitchServer.exec("ViewController", "liveview", [True])
>>> image = imswitchServer.get_image()
>>> image
array([[-1.10142425, -1.92491431, -1.20578026, ...,  0.06197434,
         2.58134274,  1.5358419 ],
       [ 0.42901765,  1.44121983, -1.62533404, ..., -0.85819345,
        -0.55907752, -1.44523978],
       [-1.89064569, -0.0806102 , -1.20610543, ..., -0.19431504,
         0.40223848,  0.78403648],
       ...,
       [-0.52828162, -1.15186004, -0.42286017, ...,  0.10683187,
         1.30652227,  0.21429759],
       [ 0.41525955,  1.69520429, -0.42840672, ...,  1.38124483,
         0.37169698,  0.46900387],
       [ 3.71865584,  2.19155528,  0.24225213, ..., -0.8929803 ,
        -0.51068475,  2.80787008]])
```

kasasxav, 2022-03-10:
Great 👏🏼👏🏼👏🏼

closed 2022-03-10 by kasasxav