ImSwitch/ImSwitch #115: Adding frequency modulation widgets for lasers

jacopoabramo, 2022-03-14:
Greetings,

I'm currently trying to upgrade ImSwitch to work with a laser which supports frequency modulation. We would like to control both the frequency and the duty cycle of our laser and for that end I'm trying to add:

- a checkbox to enable/disable the frequency modulation;
- a line edit to change the frequency (or a spin box with minimum/maximum range);
- a spin box to change the duty cycle.

I was able to find the `LaserWidget` class which I presume handles all the laser modules in the GUI, but my first attempts were a bit disappointing because I don't exactly understand how the grid for the laser widget is structured.

What I would like to get is something like this:
[image]
Of course all the logic should be adapted to make sure that a laser can be set with a frequency modulation (I'm thinking of adding a parameter in the `LaserManager` similar to `isBinary`).

Could I get some insights into this? Or do you think another approach could be better?

Thanks!

kasasxav, 2022-03-14:
Hi!

So, as you suggested, `LaserWidget` handles all laser modules. Each laser module is an object of the class `LaserModule`. I think you should add your elements in [lines 250-255](https://github.com/kasasxav/ImSwitch/blob/master/imswitch/imcontrol/view/widgets/LaserWidget.py). Right now the `powerGrid` has two columns: one for `setPointLabel`, `minpower`,` slider`,` maxpower`; and another one for `setPointEdit` and `valueUnits`. I guess you want to add Enable modulation, frequency and Duty cycle in another column, so you would have to write: `self.powerGrid.addWidget(self.enableModulatoinEdit, 3, 0, 2, 1)`, `self.powerGrid.addWidget(self.modulationCheck, 3, 1, 1, 1)` (row, column, size_r, size_c) and so on. I like the idea of only creating those widget elements if the Laser is "modulable", so makes sense to have a parameter in `LaserManager`.

That would mean that you can have that as a parameter to the function `addLaser` in `LaserWidget`, and that function is called from `LaserController`, lines [19-23](https://github.com/kasasxav/ImSwitch/blob/master/imswitch/imcontrol/controller/controllers/LaserController.py), and there you can fetch it from its `lManager`. 

I hope this is clear enough, let me know otherwise.

jacopoabramo, 2022-03-15:
Hi,

so this is my first attempt:

[image]
```python
        # laser modulation widgets
        self.modulationEnable = QtWidgets.QCheckBox("Enable")
        self.modulationFrequencyLabel = QtWidgets.QLabel("Frequency")
        self.modulationFrequencyLineEdit = QtWidgets.QLineEdit()
        self.modulationDutyCycleLabel = QtWidgets.QLabel("Duty cycle")
        self.modulationDutyCycleSpinBox = QtWidgets.QSpinBox()
        self.modulationDutyCycleSpinBox.setRange(1, 100)
        self.modulationDutyCycleSpinBox.setValue(50)
        self.modulationGroup = QtWidgets.QGroupBox("Laser modulation")
        self.modulationLayout = QtWidgets.QHBoxLayout()
        self.modulationLayout.addWidget(self.modulationEnable)
        self.modulationLayout.addWidget(self.modulationFrequencyLabel)
        self.modulationLayout.addWidget(self.modulationFrequencyLineEdit)
        self.modulationLayout.addWidget(self.modulationDutyCycleLabel)
        self.modulationLayout.addWidget(self.modulationDutyCycleSpinBox)
        self.modulationGroup.setLayout(self.modulationLayout)

        self.powerGrid.addWidget(self.modulationGroup, 3, 3, 4, 1)
```

The image is an excerpt of the CoolLED laser example setup. If I pick only a single laser I get the following:

[image]

It's a bit stretched but considering that hardly people will use only the laser widget to control their setups I'm okay with it. I had to use a QGroupBox because every time I attempted the widgets singularly and I only obtained a complete mess with the slider (it would have been compressed on the right side, at least in this way I can keep consistency).

Does it look ok? I'm not particularly happy with how the frequency line edit stretches, so any suggestion is welcome.

jacopoabramo, 2022-03-15:
Ok with the `BetterSlider` it looks much better. I changed the checkbox and used the `BetterPushButton` from `guitools`, and added the frequency range. This is the final result:

[image]

I'll keep it like this and start to implement the logic.

kasasxav, 2022-03-24:
#118

closed 2022-03-24 by kasasxav