ImSwitch/ImSwitch #117: Having @APIExport() in functions outside Controllers?

beniroquai, 2022-03-16:
Hey, looking into the `api.py` class, I think its not possible to have functions outside the Controllers (e.g. inside Managers), hooked up to the API definition, right? Is there any way to bypass that? E.g. directly controlling a hardware feature without going too deep into the code. 

Thanks!

kasasxav, 2022-03-17:
Hi Benedict!

Actually it was a bit of a decision because we thought that then it could be a bit messy to directly access the model layer. But usually what we do is to create a function in the controller for taking care of it. It's kind of the idea behind structuring the code with the model-view-controller pattern actually. It could be an issue to control the hardware from different places so I like the idea of having the controller take care of it, what do you think? If you check for example the [`PositionerController`](https://github.com/kasasxav/ImSwitch/blob/master/imswitch/imcontrol/controller/controllers/PositionerController.py) we have there functions to interact with the positioners, but then it does it using the `PositionersManager`, which has a list of `PositionerManager`. If this has any practical limitation for you we could think of alternatives, maybe creating some module in the controller to take care of this or enabling the [`MasterController`](https://github.com/kasasxav/ImSwitch/blob/master/imswitch/imcontrol/controller/MasterController.py) (which has the managers inside) to have api exported functions.

beniroquai, 2022-03-18:
I think this makes totally sense. It is just tedious to write the Controller-Manager structure for every new function that does not fit in the current framework ;-) 
Ideally there is a template from which you can quickly create new hardware and "snap" it in; like a plugin or extension. Right now I have to remember three (?) different places to add the different managers and controllers in order to get it working. Having a folder, where I place a new device and ImSwitch automatically registers it would be a dream - but I understand that this is quite an effort to realize.

kasasxav, 2022-03-18:
Yes I understand, I will keep it mind. It's a good idea :) Unfortunately, I have limited time and I have to prioritize other tasks :l

closed 2022-03-18 by kasasxav