ImSwitch/ImSwitch #120: Storing relative position of stages

jacopoabramo, 2022-03-24:
Greetings,

I'm currently adapting the Mad City Labs stages into ImSwitch. So far I was able to make them work, now what I want to achieve is:

- add a calibration routine as a method for all `PositionerManagers` (in the same way I did for the laser frequency modulation);
- setting a relative-zero position of the stages
- storing the current position in the ImSwitch settings

While for the first problem I already know what to do, for the second and third problem I don't know exactly what to do, not as much in terms of GUI adaptation (it would mean adding a couple of buttons and nothing more), but more in terms of how to store positions in the same JSON file for the hardware control configuration. Would it be possible somehow? I'm assuming it would be just easy as to read the JSON file, changing some fields, and then saving it again. I just need a tip on where into ImSwitch doing this operation.

jacopoabramo, 2022-03-24:
Note: what I mean with calibration routine is to find the _absolute_ zero position of my stages (what is considered 0 in terms of each X, Y and Z stage).

kasasxav, 2022-03-25:
Hi,
And the position you want to store in the json file is only one per stage? or kind of different values for different settings? I think I didn't get that part. And what does that value depend on / how often does it have to be changed?

jacopoabramo, 2022-03-25:
Hi,

yes it would be one per stage. In my case, I have a 3-axis stage (XYZ). I would like to have something like this:

```
startingPosition: {
    "X" : 10 # micrometers
    "Y" : 20 # micrometers
    "Z" : 25 # micrometers
}
```
So a use case for each button would be the following:
- I read the starting position from the JSON file and start moving the stage around; at some point I want to save the current position, press the "Store position" button and update this section of the JSON file accordingly.
- I want to set the current position of my stage as zero for all axis, I press the "Set zero" and update the axis position in the JSON file as zero. Also, the widget showing the stage position also is set to zero.
- I want to set the stages to their true zero position, I press the "Calibrate" button, the stages do the relative routine written by the user, the stage goes to the absolute zero position in respect of their size and I write in the JSON file - again - zero for all axis.

Now everytime ImSwitch is started, the stages should read the starting position and set the widgets accordingly. On the manager side, I think this is pretty easy because the `PositionerManager` class already provides the dictionary `initialPosition`, but so far in all the managers I checked it has been always initialized to zero for all axis. What I'm missing is how to update the JSON file without making a mess in reading/writing the file.

I hope it's a bit more clear now.

jacopoabramo, 2022-03-25:
Of course, the Ideal situation would be to have the `startingPosition` available for all managers but I can understand it could be a pain to adapt to all other managers.

kasasxav, 2022-04-29:
Hi @jacopoabramo , I forgot about this sorry. Did you manage in the end? or do you have any question on how to do it?

kasasxav, 2022-05-11:
Hi, sorry again for the delay.
We save both the detector ROIS and the laser presets in the config file, so you can check how we do it in either [SettingsController](https://github.com/kasasxav/ImSwitch/blob/master/imswitch/imcontrol/controller/controllers/SettingsController.py) and [LasersController](https://github.com/kasasxav/ImSwitch/blob/master/imswitch/imcontrol/controller/controllers/LaserController.py)

But we usually rely on the [configfiletools](https://github.com/kasasxav/ImSwitch/blob/master/imswitch/imcontrol/model/configfiletools.py) to do so. And we usually get the config file as a parameter and then save it again (kind of rewriting it) using `saveSetupInfo`

closed 2022-05-19 by kasasxav