ImSwitch/ImSwitch #121: Possible to save frame seuqences as Movies?

beniroquai, 2022-03-25:
Hey, when storing lots of data there might be some compression necessary. I had a look into the RecordingManager, but I did not really get how I could e.g. save the frames as TIF or even applying some compression. 
Can you guide me :)

kasasxav, 2022-03-25:
Hey,
Actually now it is only possible to store a set of frames as hdf5, and when using a single snap you can choose between hdf5 and tiff. I have to say it's something that I think could definitely be improved, for example in the GUI instead of having snap format having it work for both snaps and recs (also @jacopoabramo asked for it in #98), but I really didn't have time to do that. 

I will try to facilitate it a bit for you in case you want to go with it. I created a new branch [recording-improvements](https://github.com/kasasxav/ImSwitch/tree/recording-improvements). The steps to follow would be to: 

- Change the text of the "Snap format" widget to just "Format" -> [f928cc7](https://github.com/kasasxav/ImSwitch/commit/f928cc7b64225a88a997a527bc9575a96d187f23)
- Add "saveFormat" in recording controller & recording manager -> [d44e313](https://github.com/kasasxav/ImSwitch/commit/d44e31325df566bc5e7ef392cc7f54a7db9cf87e)
- Now you can add in the recording worker a block of code to save the images into TIFF, or you can also add more possibilities like TIFFCompressed. Around line [215](https://github.com/kasasxav/ImSwitch/blob/recording-improvements/imswitch/imcontrol/model/managers/RecordingManager.py) you can add an if self.snapFormat == 'TIFF' for example.

beniroquai, 2022-03-26:
Just so that I get it right, the mechanism works such that you first fill up the RAM and then write it to the disk?! Isn't this a bad way of dealing with limited Memory? :D 
Just asking. Perhaps I'm understanding the code [here](https://github.com/kasasxav/ImSwitch/blob/recording-improvements/imswitch/imcontrol/model/managers/RecordingManager.py#L304) in a wrong way. @stafak 

Couldn't we write frames to the HDF5 dataset continuously while chunks are drawn from the buffer?

stafak, 2022-03-26:
Hey @beniroquai !

It's been a while since I worked on this, but from what I remember, it works like this:
* In `SaveMode.Disk`, the data is only written to the disk. The h5py library buffers the HDF5 data to the disk without filling up the RAM.
* In `SaveMode.RAM`, all of the data is written to the RAM. It uses `BytesIO` objects, which are file-like objects which fully store data in the RAM.
* `SaveMode.DiskAndRAM` is mostly the same as `SaveMode.Disk`, except the file is not closed.

Unless you've enabled the image reconstruction module in ImSwitch, the save mode for recordings will always be `SaveMode.Disk` (assuming this hasn't changed since I worked on ImSwitch).

The code you linked is pretty badly commented (hmm I wonder who's fault it is 😅), but maybe this clears things up a bit:

```python
                if self.saveMode == SaveMode.RAM or self.saveMode == SaveMode.DiskAndRAM:
                    filePath = filePaths[detectorName]
                    name = os.path.basename(filePath)
                    if self.saveMode == SaveMode.RAM:
                        # Handle RAM-only recordings
                        file.close()  # This is a RAM-only file - it does not exist on the disk
                        self.__recordingManager.sigMemoryRecordingAvailable.emit(
                            name, fileDests[detectorName], filePath, False
                        )
                    else:
                        # Handle recordings to disk that should not be closed
                        file.flush()  # Flush the HDF5 buffer to disk (to ensure the file is fully written)
                        self.__recordingManager.sigMemoryRecordingAvailable.emit(
                            name, file, filePath, True
                        )
                else:
                    # Handle disk-only recordings
                    file.close()  # Close the HDF5 file
```

As for the issue itself, it's still possible that compression could help, if the memory is filling up because the disk can't keep up. From what I understand, you can enable HDF5 compression by passing a `compression` argument to [`create_dataset`](https://github.com/kasasxav/ImSwitch/blob/d44e31325df566bc5e7ef392cc7f54a7db9cf87e/imswitch/imcontrol/model/managers/RecordingManager.py#L194), as outlined in the [h5py documentation](https://docs.h5py.org/en/stable/high/dataset.html#filter-pipeline).

Adding TIFF suppport for recordings is more difficult, since we wrote the recording functionality around the HDF5 format originally. I think a good solution would involve abstraction of the file I/O operations – `RecordingWorker` should be free of HDF5-specific operations. But I'm not sure if saving recordings in TIFF format would help with the problem you're experiencing.

beniroquai, 2022-03-26:
Ah, thanks for clarification. I guess I need to better understands of `datasets` in the code. Does it act as a "pointer"? I couldn't find the link between `file` and e.g. `newframe`, but perhaps it's "just" the HDF5 object that does the job. 

For TIFF, I (brute-force) implemented it [here](https://github.com/beniroquai/ImSwitch/blob/record_video/imswitch/imcontrol/model/managers/RecordingManager.py#L297). Not optimized for Fiji or anything. No compression.. One More thing I would like to do is implementing a videowriter - yup. Who cares about rawdata :D .. I'm collecting way too much data to not try it out 🤩

stafak, 2022-03-26:
> Ah, thanks for clarification. I guess I need to better understands of `datasets` in the code. Does it act as a "pointer"? I couldn't find the link between `file` and e.g. `newframe`, but perhaps it's "just" the HDF5 object that does the job.

Are you talking about operations like `dataset[it:it + n, :, :] = newFrames`? If so, then I don't remember a lot about how h5py library works unfortunately, but I know that it does a lot of things behind the scenes. My **guess** is that while it looks like you're writing to a regular array, in reality this huge array doesn't exist and you're actually writing to a memory buffer, and when the buffer is full it is automatically written to the disk.

> For TIFF, I (brute-force) implemented it [here](https://github.com/beniroquai/ImSwitch/blob/record_video/imswitch/imcontrol/model/managers/RecordingManager.py#L297). Not optimized for Fiji or [...]

Haha nice :D Did it work better than HDF5?

beniroquai, 2022-03-27:
Now: Writing as Video works, too. 
Closing the issue for now.
@kasasxav perhaps it'S worth thinking about another PR? 
All in [Master](https://github.com/beniroquai/ImSwitch/)

closed 2022-03-27 by beniroquai

kasasxav, 2022-03-27:
Thank you @beniroquai and @stafak 🙂 And yes, as far I as understood the indexing in the hdf5 case to the variable dataset allows you to copy the data directly in bunches and works as a pointer and like Staffan said.

kasasxav, 2022-03-27:
And sounds great with a PR, maybe we can discuss about this during the week?