ImSwitch/ImSwitch #126: Image not showing in Napari in Windows 7 (yup, still running 🙃)

beniroquai, 2022-05-11:
**Bug description**
Visualizing a livestream of a PCO camera does not render on the big screen in napari but in the preview snippet. Any idea if this is a related PyQT-win7 issue? How to debug? Also the random-pattern fallback/simulation does not render.

**How to reproduce**
Steps to reproduce the behavior:
1. start ImSwtich with the latest fork from the UC2 branch: https://github.com/beniroquai/ImSwitch/tree/b8eebc524df2d6a311613621e1ecf2848fc88a08
2. Starting camea live stream

**Expected behavior**
The camera stream gets rendered in the Napari window

**Screenshots**
[image]

**Environment**
 - OS: Windows 7, Anaconda 3.8, Python 3.8
 - ImSwitch version: [[e.g. "1.1.0", or a branch name or commit hash for development versions]](https://github.com/beniroquai/ImSwitch/tree/b8eebc524df2d6a311613621e1ecf2848fc88a08)

**Additional context**
-

beniroquai, 2022-05-11:
Maybe related: 
https://github.com/napari/napari/issues/1981

Meaning, Napari does not work on Windows 7 ;-)

kasasxav, 2022-05-11:
Hey,

Did you try if the same code/config file/hardware works with windows 10?

kasasxav, 2022-05-11:
something you can try is to install the latest version of napari in your environment: `pip install napari==0.4.15`. Our version of napari is quite old because of an old issue that I think now it is fixed.

beniroquai, 2022-05-11:
Thanks! The code works on Win10 (just tested). Will try to update napari. Any reason why ImSwitch uses an older version?