ImSwitch/ImSwitch #130: Weird widget visualization

jacopoabramo, 2022-05-25:
Greetings,

I've merged the most recent changes in my repo. When loading a configuration with a positioner widget, I'm getting this weird visualization:

[image]

I'm thinking that the problem is in the `PositionerWidget` class, specifically line [64](https://github.com/kasasxav/ImSwitch/blob/4a03621b2705a0270e6b8c43e4083a656a5553b7/imswitch/imcontrol/view/widgets/PositionerWidget.py#L64), which creates this effect. I'm not entirely sure how to fix this. For now I commented it out since I don't use the speed control for my positioners.

kasasxav, 2022-06-10:
[Here](https://github.com/kasasxav/ImSwitch/commit/bb50dd6238f3b8cac4d7c1f77824415dbe3e4657) and [here](https://github.com/kasasxav/ImSwitch/commit/88ad5ea5a006f27bb108dbd0aa56dd3d202e5c2c) is how I did the speed optional. Is everything fixed then?

jacopoabramo, 2022-06-10:
I'll merge when I have some time and test it on my fork. Meanwhile I was actually hoping if you could also check out the default step size of the positioners (if you haven't done so already). Having a 1000 step size with my stages is a bit risky and I would want to avoid and instead having a smaller default step size or the possibility to set my own. I can open a new issue if you so prefer.

kasasxav, 2022-06-10:
I agree, I think now it should be fixed as well. I think it was because of the uc2 steppers. In my case it's also risky - in the future we could have that in the config file.

kasasxav, 2022-10-16:
I close this because I tested myself and everthing seems fine

commit b7be3807 (2022-06-10): Fix #130
closed 2022-10-16 by kasasxav