ImSwitch/ImSwitch #131: Failing to do HDF5 recordings

jacopoabramo, 2022-05-25:
Greetings,

after merging the changes from the your repo to my fork I tried to do a recording with HDF5 (number frames mode). This is what I got from the stack trace:

```python
File "C:\Users\iScat_Lab\AppData\Local\Programs\Python\Python39\lib\site-packages\imswitch\imcontrol\model\managers\RecordingManager.py", line 225, in run
    self._record()
  File "C:\Users\iScat_Lab\AppData\Local\Programs\Python\Python39\lib\site-packages\imswitch\imcontrol\model\managers\RecordingManager.py", line 275, in _record
    datasets[detectorName] = files[detectorName].create_dataset(
  File "C:\Users\iScat_Lab\AppData\Local\Programs\Python\Python39\lib\site-packages\h5py\_hl\group.py", line 149, in create_dataset
    dsid = dataset.make_new_dset(group, shape, dtype, data, name, **kwds)
  File "C:\Users\iScat_Lab\AppData\Local\Programs\Python\Python39\lib\site-packages\h5py\_hl\dataset.py", line 140, in make_new_dset
    dset_id = h5d.create(parent.id, name, tid, sid, dcpl=dcpl)
  File "h5py\_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
  File "h5py\_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
  File "h5py\h5d.pyx", line 87, in h5py.h5d.create
ValueError: Unable to create dataset (name already exists)
```

I honestly have no idea what happens here. Any input appreciated!

beniroquai, 2022-05-26:
I fixed the error in the @beniroquai-fork: https://github.com/beniroquai/ImSwitch. It's most likely a copy-paste error in the `RecordingManager.py`

See [here](https://github.com/beniroquai/ImSwitch/commit/3724383e880917293779d876ccea577217ea74b3). 

Should I add a PR @kasasxav ?

kasasxav, 2022-06-08:
Yes sure!

kasasxav, 2022-06-10:
Can you check if it works now?

jacopoabramo, 2022-06-10:
I'll merge and report here!

kasasxav, 2022-10-16:
Was this issue solved? :)

jacopoabramo, 2022-10-16:
Yup, I have to admit I don't remember exactly how because some time is passed. I'll close the issue anyway.

closed 2022-10-16 by jacopoabramo