ImSwitch/ImSwitch #133: Saving in TIFF format causes recording to be stored only as a single frame

jacopoabramo, 2022-06-09:
Greetings,

I have a small problem. When trying to record a video in TIFF format and reading it back via napari by opening it in the viewer, it appears that only 1 frame is stored.
This does not happen in stead in ImageJ as the full stack appears and can be slided through.

I understand that you use the tifffile package to write the single images into the recordings, I don't know if this issue has already appeared.

I'll try to give more details if needed.

Thanks.

kasasxav, 2022-06-10:
I think @beniroquai was the one that did the TIFF option for saving stacks, right?

jacopoabramo, 2022-06-10:
Yes I recall that he was working on that. I thought that the `append` option was causing the issue, but even after removing it it didn't change much. What I'm thinking is happening is that the manager is writing multiple pages on the same file, and ImageJ is capable of parsing them correctly, while instead napari has some issues because usually it treats TIFF as a single page with multiple frames.

kasasxav, 2022-06-10:
Can you check now just in case? I uploaded his recordingManager to test.

kasasxav, 2022-06-10:
ah right.. that makes sense... Could it be the ImageJ option when saving tiffs?

jacopoabramo, 2022-06-10:
I don't think so, the ImageJ option is only used by the HDF5 file saver, the TIFF saver doesn't use its parameters. Currently the TIFF writer just blindly streams images into the saving file.

jacopoabramo, 2022-06-10:
To be honest the best approach here would be to probably refactor the RecordingWorker class. It's just too much spaghetti code and debugging becomes time consuming.

kasasxav, 2022-06-10:
I agree, the recording stuff is a bit of a mess, we should sit down and work on it at some point.

jacopoabramo, 2022-06-10:
I think I found the correct way on how to do it the way we want it. There's a snippet example [here](https://github.com/cgohlke/tifffile):

Successively write the frames of one contiguous series to a TIFF file:

```python
>>> data = numpy.random.randint(0, 255, (30, 301, 219), 'uint8')
>>> with TiffWriter('temp.tif') as tif:
...     for frame in data:
...         tif.write(frame, contiguous=True)
```

I just tested this out with another piece of code and it works just fine.

beniroquai, 2022-06-10:
Interesting. Would this file open in Fiji assuming it is >4GB? 

I agree, the RecordingManager is really hard to read with all the different cases. Formats available now: HDF5, TIF (multipage and many single files) and MP4 (yes indeed ;-) )

jacopoabramo, 2022-06-11:
> Interesting. Would this file open in Fiji assuming it is >4GB?

I think that if you set the flag `imagej=True` or `bigtiff=True` in the `TiffWriter` constructor it should work. I haven't had the chance to test it though. I remember that they are maybe mutually exclusive so it's best if you check the API documentation.

kasasxav, 2022-10-16:
I close this for now but hopefully soon we can improve the recording controlls!

closed 2022-10-16 by kasasxav