ImSwitch/ImSwitch #134: Z-Stage out of focus at software restart

jacopoabramo, 2022-06-14:
Greetings,

I'm dealing with a weird behavior of the stages in my iScat setup. I'm using the MMCore with pymmcore to control the X-Y-Z stages from MadCityLabs, and every time I restart ImSwitch the Z-stage goes out of focus. I was wondering if somewhere in the code there's a call to the `PositionerManager` method `setPosition` which at startup or during shutdown moves the stages to a 0-position of some sort.

kasasxav, 2022-10-16:
Hi,

I am so sorry for my delay, but there is this attribute in PositionerManager called "initialPosition":
https://github.com/kasasxav/ImSwitch/blob/master/imswitch/imcontrol/model/managers/positioners/PositionerManager.py

This is fetched from the configuration file and then initialized by PositionersManager (https://github.com/kasasxav/ImSwitch/blob/master/imswitch/imcontrol/model/managers/PositionersManager.py) 

So I think you just have to add it in the positioner info in your ImSwitchConfigFolder

jacopoabramo, 2022-10-16:
I reworked quite a bit the positioner manager for the MMCore positioner manager and now all the stage controls should be fine. I only tested with a specific device adapter but hopefully this can be exteded to other positioners as well. Also the device adapter for the Z stage used for focusing was bugged, I managed by installing a nightly build of micro-manager with an updated version of the device adapter. I'll close the issue.

closed 2022-10-16 by jacopoabramo