ImSwitch/ImSwitch #137: update levels issue

AZehrer, 2022-07-15:
Hi, 
since this week, i seem to have the problem that the update levels is not working, and giving me following error:

2022-07-13 14:23:23 ERROR [ExceptionHandler] Traceback (most recent call last):
  File "c:\users\ando\documents\imswitch_dev\imswitch\imcommon\view\guitools\naparitools.py", line 138, in _on_update_levels
    for layer in self.viewer.layers.selected:
AttributeError: 'LayerList' object has no attribute 'selected'

Can you maybe have a look at it?
Best wishes, Ando

kasasxav, 2022-10-16:
Hi,

I am sorry but I could not reproduce this issue, did you manage to fix it?

AZehrer, 2022-10-20:
Hi, sorry for the late reply. 
I don'tknow how, but i don not have the issue anymore.

closed 2022-10-21 by kasasxav