ImSwitch/ImSwitch #139: ImSwitch on Raspberry Pi, installation

flyingpinguu, 2022-10-07:
Hey there,

I'm looking to install ImSwitch on Raspian on a Raspberry Pi 4 for a flourescence microscope build based on UC2. I want to use it to controll the stepper motors and visualize the pi camera. The problem is that installation with pip does not work, I get many error messages and in the end it says there is some conflict with QScintilla (Terminal code is here [https://gist.github.com/flyingpinguu/6e917a4034b77461140bc9f2c7058cc2](url)). I also tried getting the UC2 version, but a conda environment is needed. I couldn't find one for Raspian.
Do you think there is a not too difficult solution to my problem?

Thanks a lot,
Alexander

kasasxav, 2022-10-07:
I think @beniroquai had the same issue, somehow the QScintilla was giving problems with the raspberry pi. You could try to go to the confic folder (should be Documents/ImSwitchConfig/config/modules) and remove imscripting from the list. Let me know if that works.

flyingpinguu, 2022-10-07:
Hm, I'm not sure where I can find the directory you mentioned. I can download and unzip the git, but it seems to be structured differently.

kasasxav, 2022-10-07:
Hi, 

So usually it appears when you install ImSwitch. You can go to the folder of the github and write "pip install .".

Or whenever you run ImSwitch for the first time that will create a folder in the place I told you (Documents/ImSwitchConfig)

flyingpinguu, 2022-10-07:
Ah okey, thanks again. The problem is, that it doesn't install Imswitch in the first place. I tried going to the folder and installing from there, but I have the same problem as with "pip install ImSwitch".

kasasxav, 2022-10-07:
mmh I see, can you try to remove QScintilla from the list of dependencies in setup.cfg and try again installing the requirements? or like pip install .

flyingpinguu, 2022-10-07:
I did, it was able to install it this time. Trying to start it wasn't possible however, it is missing a module called h5py (terminal: https://gist.github.com/flyingpinguu/adc288b6d8808b6b6afcb094816cf6a4). Also I got a lot of warnings regarding other modules, that they are in directories which are not "on PATH". Not sure if that's bad.

beniroquai, 2022-10-11:
Hey, I can try to assist. However, we installed it on the jetson Nano. The
tricky but was getting qt working

kasasxav, 2022-10-16:
@flyingpinguu could you try this: https://stackoverflow.com/questions/62433716/how-to-solve-importerror-libhdf5-serial-so-103-cannot-open-shared-object-file ?

Seems like other people experienced the same problem as you!

kasasxav, 2022-10-25:
how is it going @flyingpinguu ? Did you manage?

indev-aj, 2022-10-25:
> 

Hi @beniroquai I'm trying to install imswitch on jetson nano, however I encountered a problem that was due to pyqt (i believe)
here's the [error from terminal](https://imgur.com/a/RSPC6mf)

Do you have workaround for this? Thank you.

indev-aj, 2022-10-25:
I tried installing pyqt using `pip3.9 install pyqt5` but there was some kind of error, didn't screenshot though. That was the only command I did for installing the pyqt.

Note I'm using python 3.9.15

beniroquai, 2022-10-25:
@indev-aj  If I remember correctly, I had to create a virtual environment to install it through miniconda. 
It'S almost a year ago that I have started it and right now it'S "just" running. 
I put down a (non-complete) list of installation instructions here:
https://github.com/openUC2/ImSwitch#on-jetson-nano

Let me know if this is of help.

indev-aj, 2022-10-27:
@beniroquai I've tried installing imswitch through the steps but then I go an error with QScintilla. I commented out qscintilla from cfg, and now I got an error `No matching distribution found for pywin32>=304`

indev-aj, 2022-10-27:
> @beniroquai I've tried installing imswitch through the steps but then I go an error with QScintilla. I commented out qscintilla from cfg, and now I got an error `No matching distribution found for pywin32>=304`

managed to solve this, and able to run imswitch however, I face two errors out of the bat.
`ModuleNotFoundError: No module named 'imswitch.imcontrol.model.managers.positioners.NidaqPositionerManager'` when running the examples.

I also got `ImportError: cannot import name 'Qsci' from 'PyQt5'` 
Whenever I try to install qsci, the binding never succeed and [showed this instead](https://imgur.com/a/41dxPfJ)

beniroquai, 2022-10-27:
Hey, great news. I skipped using Nidaq since this gives weird errors on
multiple machines. Sorry.
Qsci is not working indeed. You can remove this from the config.json
instead.

kasasxav, 2022-10-27:
I remember you had an issue with QScintilla indeed, could you send me the error message?

indev-aj, 2022-11-07:
@beniroquai after some breaks and teas, I managed to run imswitch (except for the scripting because of qsci) thanks for the help :)

beniroquai, 2022-11-07:
Thanks for reporting! Were the steps I mentioned enough to install
Imswitch? Should I add additional notes?

> Message ID: ***@***.***>
>

indev-aj, 2022-11-07:
Not really sure if there's anything to add, it just that I tried to just follow the "On Jetson" so I forgot to use the provided samples for UC lol

beniroquai, 2022-11-07:
No worries. but if this is the way to install ImSwitch on the Jetson. I'm
happy :)

indev-aj, 2022-11-09:
Hello again, @beniroquai not sure if I should continue in this thread or not. According to the samples from the uc2 version, there is manager for thorcam. I have a DCxx series camera, but I can't install the driver for it on jetson. If I use `lsusb` I'd get `ids imaging development systems gmbh generic (firmware not loaded yet)` 

Any idea for the driver?

beniroquai, 2022-11-09:
@indev-aj, I think you could open a new thread. Interesting though, that the Thorcam is actually an IDS camera. Didn't expect that. The IDS should have linux ARM drivers if I'm not mistaking. But I'm not very familiar with them. It wasn't too easy to get the old ones working on Windows..

Soyalexf, 2022-12-07:
@flyingpinguu hey bro can you install it on your raspberry? If it is, can you explain to me how?

beniroquai, 2022-12-07:
We should also update the documentation. Did you follow the readme in the
openuc2 repo?

kasasxav, 2023-01-16:
how is this going? @Soyalexf @indev-aj @flyingpinguu

beniroquai, 2023-01-16:
We are facing similar installation issues on the MacBook Pro M1 which is
also an ARM64 similar to the Jetson. We got it working on both now :)

> Message ID: ***@***.***>
>

closed 2023-01-29 by kasasxav