ImSwitch/ImSwitch #141: `pyqtgraph.ptime` module is deprecated starting version 0.13.1

jacopoabramo, 2022-10-20:
I recently changed the workstation connected to my instrument and re-installed Python (version 3.8.10) and all the ImSwitch dependencies.

I stumbled across an import error of `pyqtgraph.ptime`; I dug a bit and found the following [pr](https://github.com/pyqtgraph/pyqtgraph/commit/d396d3379968331bcd2bf09f55ff481b292d2e40)

So far I managed to fix the error by fixing pyqtgraph to version 0.12.1 in the dependencies, but I would reccomend adapting this based on the Python version the user has.

kasasxav, 2022-10-25:
Thank you Jacopo! I can just subtsitute by the ref_counter right? No matter the version I guess.

jacopoabramo, 2022-10-25:
From version 3.7+ only perf_counter is supported. Maybe you should consider making a check when versions below that are used for ImSwitch - unless version 3.6 is for example not supported at all.

kasasxav, 2022-10-25:
Yes but my point is that if I just change to perf_counter everything will be supported, right?

jacopoabramo, 2022-10-25:
Yes this should fix the problem. I'll close the issue.

closed 2022-10-25 by jacopoabramo