ImSwitch/ImSwitch #143: PhotometricsManager gets stuck while recording

jacopoabramo, 2022-11-22:
Greetings,

With another workstation we had some issues in using the PhotometricsManager for recording. From what I understood, the manager uses a try-exception approach to store the frames in a local list. In our case the exception never occurred, and what happened is that the RAM usage would explode and ImSwitch crashed.

Right now I changed the approach adding a number parameter to set the maximum size of the chunk to be transferred while recording and iterate via a for loop:

```python
    def getChunk(self):
        chunkFrames = [] 
        status = self._camera.check_frame_status()
        try:
            if not status == "READOUT_NOT_ACTIVE":
                for _ in range(len(self.__chunkFramesSize)):
                    im = np.array(self._camera.poll_frame()[0]['pixel_data'])
                    chunkFrames.append(im)
        except RuntimeError:
            pass
        return chunkFrames
```

Maybe this is not the ideal approach but at least now it works. I believe though that using the capture sequence API provided by Photometrics should be the better approach.

kasasxav, 2022-11-25:
Hi Jacopo,

Thank you ! Yes actually, I have to say that I only had the Photometrics for a week so I didn't have much time to develop this. Feel free to open a PR!

kasasxav, 2023-01-29:
Hi @jacopoabramo , did you manage to fix it?

jacopoabramo, 2023-01-30:
Yes I managed to fix it. I'll close the issue

closed 2023-01-30 by jacopoabramo