ImSwitch/ImSwitch #145: First 100 frames look the same when acquiring a stack in .hdf5

AZehrer, 2023-01-12:
**Bug description**
When acquiring a stack of images e.g. 1000 frames, the first 100 frames seem to be exactly the same. It happens at different exposure times (10ms-50ms). This problem seems also to only happen when using .hdf5 and not .tiff.

**How to reproduce**
Steps to reproduce the behavior:
1. Acquire 1000 frames of a sample with 25ms exposure time of a moving sample
2. See the motion is not present in the first 100 frames
3.

**Expected behavior**
The frames after the first 100 frames are normal. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment**
 - OS: [e.g. "Windows 10"]
 - ImSwitch version: [e.g. "1.1.0", or a branch name or commit hash for development versions]
 - How you downloaded ImSwitch: [either "pip" or "standalone bundle"]

**Additional context**
Add any other context about the problem here, if applicable.

kasasxav, 2023-01-12:
Which camera are you using? I think the problem should be in the camera manager

AZehrer, 2023-01-12:
I am using the Alvium 1800 C-507 CMOS, from allied vision.

kasasxav, 2023-01-12:
Alright! Thanks for reporting the bug btw. I think this manager I have not done it myself so I'm not sure, maybe it's from the UC2 repo? Can you take a look at getChunk and make sure that it's returning the new list of frames, in the camera manager? getChunk is the one that is called from the RecordingManager so that would be the one that is failing in my opinion

AZehrer, 2023-01-12:
Thanks for the quick feedback. I will look into that

kasasxav, 2023-01-29:
I close here since you added one in openUC2, hope you managed!

closed 2023-01-29 by kasasxav