ImSwitch/ImSwitch #160: Where to add tests?

dboonz, 2023-04-20:
I am developing support for smarACT stages and I'd like to write a few tests. It's nothing fancy, but for instance trying to connect to the stage and checkingthe serial number, or changing a setting on the stage and checking that that's properly done. For this kind of stuff it typically makes sense to add some unit tests. For this, it's nice not to have to start up imswitch which can take quite a while. After some trial and error I found a way to do this (at the bottom).

I basically have two questions:

1. Is this the right way to do it?
2. If so, where should I add this?

Before I file the pull request, I'll ask smaract to take a look at the wrapper, that we're not adding something that ruins someone's stage.

```python
import numpy as np

from imswitch.imcontrol.model import configfiletools

from imswitch.imcontrol.model.managers.positioners.SmarACTPositionerManager import SmarACTPositionerManager
from unittest import TestCase
import logging

from imswitch.imcontrol.view import ViewSetupInfo

logger = logging.getLogger('test stage')
logging.basicConfig(level=logging.INFO)

OK_to_run=True

class TestSmarACTPositionManager(TestCase):

    def setUp(self) -> None:
        global OK_to_run
        if OK_to_run is None:
            print(
                'WARNING!!! This is a hardware test. The stage will move in random directions. Only run this if that is safe')
            answer = input('Continue? [n]')
            if answer.lower() != 'y':
                OK_to_run = False
            else:
                OK_to_run = True
        elif OK_to_run is False:
            raise RuntimeError('Cannot run test as running hasnt been confirmed')
        options, optionsDidNotExist = configfiletools.loadOptions()
        setupInfo = configfiletools.loadSetupInfo(options, ViewSetupInfo)
        p_info = setupInfo.positioners['SmarACT']
        self.drive: SmarACTPositionerManager = SmarACTPositionerManager(positionerInfo=p_info, name='something')

    def tearDown(self) -> None:
        self.drive.finalize()

````

kasasxav, 2023-06-22:
Hi @dboonz , did you manage?
Normally they are in the folder _tests , but I don't know if it would make sense with hardware, since you would need to have it on to test it and in Github it would fail. I guess for that the best is to have a test that first tries to initialize the real device and otherwise initializes a mock one and still tests all its functions.

kasasxav, 2023-06-22:
imswitch/imcontrol/_test (@jacopoabramo)