ImSwitch/ImSwitch #168: Adding Micro-Manager core in the model

jacopoabramo, 2023-06-13:
There has been an attempt [here](https://github.com/Eggeling-Lab-Microscope-Software/ImSwitch/tree/pymmcore-plus) to add the Micro-Manager core as part of the model within ImSwitch. This should be brought into the main repo.
Unit tests are still missing, and testing on actual hardware should be done as well.

beniroquai, 2023-06-13:
Do you have any favorite hardware which should be working? We have PCO's and Andors. I guess the latter does not have python support yet, so worth testing?

jacopoabramo, 2023-06-14:
@beniroquai right now I use Micro-Manager mostly to controll my Mad City Labs stages. I did the interface for it and - at least with these stepper motors/piezos that I have in my setup - seems to be pretty stable. But potentially this way of implementing the MMCore supposedly works for all hardware which is already integrated as a device adapter.

I have some cameras that I can use to test the MMCore interface in my setup, but the microscope is often used to take measurements so I don't really want to create possible complications for now.

If you fancy lending me one of your PCOs for testing I could have a look at it.

jacopoabramo, 2023-06-14:
I'm confused on what you mean by documenting these use cases. The objective of the issue would be to find a cohesive interface that integrates with ImSwitch on a core level using - for now - the [`pymmcore`](https://github.com/pymmcore-plus/pymmcore-plus) bindings. What would you use pycro-manager for?

jacopoabramo, 2023-06-14:
> Oh sorry, I meant, it would be nice to get an idea of what has been implemented already - e.g. the online DPC reconstruction in our branch or the Holoprocessing.
> […](#)

Ohhh, I see. It does make sense though... but I wonder if the core and pycro-manager can co-exist on the same application, or if it makes more sense to use pycro-manager to extend the functionalities to allow widgets to be used as well...

jacopoabramo, 2023-10-08:
Update on this: #177 will now attempt to include PycroManager as a backend acquisition engine. It will not be a drop-in replacement but an alternative to it.

beniroquai, 2023-10-10:
This sounds amazing. Perhaps worth giving a flash talk in our next Meeting? I should have one on the virtual microscope ;)