ImSwitch/ImSwitch #169: Julia interface

jacopoabramo, 2023-06-13:
[Julia](https://julialang.org/) is becoming a interesting language for scientific computing. It used a JIT compiler to speed up execution which are comparable with other projects like numpy and - for some features like for loops - becomes even faster. There's been some interest on adding Julia as integral part of ImSwitch - by providing for example a computational engine for one or multiple widgets.

There are some solutions to integrate Julia code into Python; the most relevant ones are:

- [pyjulia](https://github.com/JuliaPy/pyjulia)
- [JuliaCall](https://github.com/cjdoris/PythonCall.jl)

There are some limitations in using Julia within Python; the biggest one is that calling trying to interact with Julia from different threads causes everything to crash. Although there could be a way to insert this into the Presenter layer so that the same thread interacts with Julia and thus making use of it safely.

Another possibility would be to integrate support for Julia code into imscripting. This could be a little bit thougher because one would need to find a way to expose the Python APIs currently provided into Julia - so it becomes a little bit convoluted.

beniroquai, 2023-06-13:
Adding @rainerheintzmann here. 

One more option would be the use of the rest API provided by fastapi. Julia itself would act as a client interface. Not super duper fast and loads of fun with type casting, but not much of a mess. I'm sure someone did that already

jacopoabramo, 2023-06-14:
I'm not a fan of using rest API for this kind of stuff, I would rather keep it more tightly integrated into the architecture and use these packages. rest APIs I find more comfortable to use in this "headless" mode you tried to use for your projects, this goes more into the GUI side