ImSwitch/ImSwitch #170: Packaging ImSwitch into a deliverable installer

jacopoabramo, 2023-06-13:
Right now ImSwitch is installed via pip. At some point it would be wise to create a package (i.e. an .msi file for Windows, and relevant counterparts for other major OSs) to be installed as a normal application.

jacopoabramo, 2023-06-13:
@beniroquai I think you suggested some packages for this but I can't seem to find the references to it.

kasasxav, 2023-06-13:
In the release page there is also a .zip with an executable file for windows ❤️ In the ImSwitch guidelines repo it is explained how to do that when doing new releases

jacopoabramo, 2023-06-13:
@kasasxav ops :P apologies, I forgot about that. but maybe @beniroquai meant something else then?

beniroquai, 2023-06-13:
Pyinstaller is a bit static: if there is a change to any of the packages, you have to download a >1GB file - not good. 
Pip is nice but not userfriendly. 
I have stumbled upon Conda Constructor that creates installers for all OSs. Folks from Napar apparently also uses it:
https://github.com/napari/packaging/blob/main/build_installers.py
and https://napari.org/0.4.17/developers/packaging.html
Their CLI seems pretty involved and they patched all the versions to get it nicely added to the menus and stuff -but I think the conda creator could be a nice first start. I have started working on it here, where the bat-file can even install drivers and stuff: https://github.com/beniroquai/ImSwitchInstaller/blob/main/constructor/post_install.bat

Alternative: Docker.

jacopoabramo, 2023-06-14:
I hate Docker but only because I'm very lazy in learning it... anyway, I think that changing type of packaging software wouldn't help that much unless there's some magic compression some software uses...?

This is also one of the problems with having a big codebase: the related installers will be pretty heavy. But 1 GB seems a little bit too much aswell... even Micro-Manager is not that heavy o_o I wonder why's that?