ImSwitch/ImSwitch #174: Headless mode (no-GUI)

jacopoabramo, 2023-06-14:
During one workshop made in Jena by @beniroquai, he used ImSwitch in "headless" mode (a.k.a. without the Viewer layer) in order to launch it on an embedded platform and then use fastAPI with a browser to control ImSwitch from there. I believe it was a Jetson, if I recall.

At any rate this point is interesting for me because there are some new platforms going around (like the PYNQ framework) who would benefit from this. It's not high priority but I still consider it a nice feature to scale down microscope hardware needs.

beniroquai, 2023-06-14:
Hey @jacopoabramo you can find the entry-point here: https://github.com/openUC2/ImSwitch/blob/master/imswitch/__main__.py#L12 (should have that as a json-switch somewhere in the config. It essentially avoids the GUI from launching. All managers and controllers and threads are still alive.

jacopoabramo, 2023-06-14:
And then what do you use to communicate with the other layers? I assume it's fastAPI? Did you do your own browser interface?

jacopoabramo, 2023-06-30:
I'm just gonna add this into the issue discussion for reminding me later: https://github.com/jnmaloney/WebGui

It's an example of how to use ImGui (which is a GUI framework used mostly in game development for quick and lightweight user interfaces) for a web application. Since it's compiled from C++ into webassembly it should be very fast. I'm not sure how fast it is in comparison to native JS applications, but personally having more experience in C++ than JS I'm keeping it as an option for the future.