ImSwitch/ImSwitch #175: Signal generators refactor and documentation

jacopoabramo, 2023-06-14:
Today @jonatanalvelid kindly helped me getting more familiar with the current status of the signal generators within ImSwitch. @kasasxav, I apologize for tormenting you with this but I really need to make some order in my mind about this :P so whenever you have time to review this information it would be great.

First some terminology because this got me confused a lot, so it's just a reminder for me:

- signal generator: the device manager that controls the hardware intended to generate signals which are then propagated with a specific pattern
- signal designer: the controller that manages the creation of this pattern through some parameters usually defined at a config file level

There are currently some problems and questions which should be addressed:
- there are some references within the `APDManager` to the `NidaqManager` class (for example [here](https://github.com/ImSwitch/ImSwitch/blob/b05a431320671c2af16b6db35bed6818d4826d70/imswitch/imcontrol/model/managers/detectors/APDManager.py#L276)) which should not be there. The reason being that the signal generator in the `APDManager` could also use an interface different from the `NidaqManager`. Speaking with Jonatan we've come to realize that there should be some way to address this problem, but he's also currently working on it. This is just a reminder to keep this under watch and fix this whenever somebody manages to work on this;
- in the [`SuperScanManager`](https://github.com/ImSwitch/ImSwitch/blob/b05a431320671c2af16b6db35bed6818d4826d70/imswitch/imcontrol/model/managers/ScanManagerBase.py#LL39C17-L39C17) class there's a reference to this `_parameterCompatibility` method which is both abstract and implemented; is this a typo?
- some nomenclature should be changed in this class too: 
   - [`_scanDesigner` ](https://github.com/ImSwitch/ImSwitch/blob/b05a431320671c2af16b6db35bed6818d4826d70/imswitch/imcontrol/model/managers/ScanManagerBase.py#LL27C21-L27C21)-> `_analogSignalDesigner`
   - [`_TTLCycleDesigner`](https://github.com/ImSwitch/ImSwitch/blob/b05a431320671c2af16b6db35bed6818d4826d70/imswitch/imcontrol/model/managers/ScanManagerBase.py#L28) -> `_digitalSignalDesigner`
- in the [`makeFullScan`](https://github.com/ImSwitch/ImSwitch/blob/b05a431320671c2af16b6db35bed6818d4826d70/imswitch/imcontrol/model/managers/ScanManagerBase.py#L96) method, there's a reference to the `scanInfoDict` dictionary, which Jonatan told me is only used within the APDManager and there are some calculations being done to obtain this dictionary. Is there any reason why this should not be moved to the APDManager and add some parameters in the APD managerProperties if necessary?
- the method `runScanAdvanced`, in the `ScanController` classes should probably be refactored to just a simple `runScan`, while `makeFullScan` in the `ScanManager` classes should be come something like `buildScanSignal`, unless I'm missing something;
- in all the `ScanController` classes there's a direct reference to the `nidaqManager` instance which should not be there, but instead there should be a reference to a `ScanManager` base class. Funnily enough, @kasasxav, if you remember this is what we talked about a long time ago whenever I did my `PulseStreamerManager` implementation :D

pskeshu, 2023-07-13:
hi,

I have been trying to use imswitch, which I love for the widgets and comms, but I think device layer and planning could be refactored out of the imswitch core to use best practices from third-party libraries which have more generalizable solutions. 

For instance, the device architecture of ophyd https://nsls-ii.github.io/ophyd/architecture.html encompasses all design concerns regarding devices stay there, which are accessible by the planning layer https://nsls-ii.github.io/bluesky/plans.html 

I have used this model in a personal project ([1](https://github.com/ndsystems/deepthought/blob/0c29de7f4cbcf08b080d5f752b8951892f799d4f/deepthought/devices.py#L314), [2](https://github.com/ndsystems/deepthought/blob/0c29de7f4cbcf08b080d5f752b8951892f799d4f/deepthought/microscope.py#L238)), and I feel it helps a lot from a dev perspective to be able to rapidly add devices to the overall system and work with new or existing plans by writing plans through an established standard (https://nsls-ii.github.io/bluesky/plans.html#stub-plans).

Best,
Kesavan

beniroquai, 2023-07-13:
>
> That sounds exciting!
>

Could You imagine having a dedicated low level manager that implements this
> interface in ImSwitch?
> Message ID: ***@***.***>
>

jacopoabramo, 2023-07-13:
Hi @pskeshu , thanks for pitching in. I'm not quite sure I understand what you mean when you say "refactor out the device layer". Could you elaborate more on this?

Also, is there a 

As for the links you posted I've only gave a glimpse and they present some really interesting concepts. What I'm afraid though in integrating all these abstraction layers on top of each other we may limit ourselves performance wise - especially when talking about camera acquisitions. But I may be mistaken since I'm still just reading the documentation.

Also how do I know if my device is supported? Is there a list of supported devices?

jacopoabramo, 2023-10-08:
@beniroquai to resuscitate this topic, for your stichting widget, do you think you would be able to adapt the suggestion for using ophyd as an abstraction level in cohesion with the scanning?

pskeshu, 2023-10-08:
@jacopoabramo Sorry, I missed your EDIT. Yes, largely that's what I'm trying to articulate - that device layer can be refactored as ophyd objects, which can simplify support to a large extent. In principle then, any new device can be added to this architecture, as long as follows a standard like ophyd, which is very mature and has community support. There are so many advantages to this, that might warrant a call.

jacopoabramo, 2023-10-08:
@pskeshu all good, thanks any way for replying. Indeed after your post I dwelve more in depth into ophyd and it's a perfect fit for exactly what we want to achieve on the scanning side - and not only.

@nornil this is close to the generic experiment planner we were discussing in the past, since bluesky objective is exactly to plan and customize experiments

pskeshu, 2023-10-08:
Just to elaborate, ophyd support gives access to the larger bluesky ecosystem, which makes it easy to work across platforms. It creates standards for [acquisition](https://blueskyproject.io/event-model/main/index.html), [planning](https://blueskyproject.io/bluesky/plans.html), and overall looks really neat for writing experiments with python. And GUI integration with bluesky is an active project. 

They might have insights on how to go about this. cc @tacaswell @danielballan

jacopoabramo, 2023-10-08:
Thanks for the insight, @pskeshu . One thing I'm trying to wrap my mind around is mostly how performant the bluesky engine can be when trying to integrate high speed acquisitions. To elaborate: I'm currently using an hacked version of ImSwitch to control an interferometric scattering microscope (iSCAT) with a side fluorescence channel; the iSCAT channel uses a camera capable of reaching up to 10k FPS. As we use it for particle tracking experiments with exposure times as low as 300 us, the current recording engine allows me to only record a certain number of frames before losing them. Would bluesky be able to achieve such high framerates? If so, how? One thing I've been pondering is accelerating the model by compiling it with Cython; would this be a possibility for bluesky as well?

pskeshu, 2023-10-08:
A question for the nikea community perhaps - https://join.slack.com/t/nikea/shared_invite/zt-24yl9vom8-Dac~gmCsQkflInf4ayFVNw