ImSwitch/ImSwitch #178: General behaviour of background activities (e.g. Stage movement)

beniroquai, 2023-06-17:
The way imswitch is currently working is that there are certain activites eventually blocking the main thread. Especially tasks such as moving the stage or turning on a laser, etc. I was wondering if this is worth mentioning to eventually improve and move them to different tasks/schedulers. I remember the openflexure server (https://gitlab.com/openflexure/openflexure-microscope-server) uses an interesting approach there. Since everything is handled through the REST-API they have to rely on timings and asynchrounous behaviour. 
Right now, if I trigger a motion I should rely on a callback or function that waits until the stage has arrived. I wish to have blocking functions that do not block imswitch's main thread. Is there any way to do that?

jacopoabramo, 2023-06-18:
I managed to "synchronize" camera acquisition and stage movement by using Pyro5 to remotely connect my ximea manager to the ImSwitchServer and run an acquisition routine on a separate thread. You could probably do the same since if I got it correctly the restAPI is served by the same thread

beniroquai, 2023-06-18:
Hm, but if I start a motion the napari live viewer should pause in the same way if the stage is blocking, no? 

Perhaps the keyworkd  `runOnUIThread` in `@APIExport(runOnUIThread=True)`  may change this behaviour somehow?

jacopoabramo, 2023-06-19:
No it should not, I do the exact same thing you're suggesting:

- I open a Pyro5 proxy in the `XimeaManager` thread;
- from this proxy I send a request to the `ImSwitchServer` Pyro5 instance which:
  - starts moving the stage over a series of steps in a square movement;
  - after each movement collects an image;
  - sends a signal using the communication channel to store the processed median of the collected stack in the `XimeaManager`

This type of operation should not be blocking, because everything is handled by the `ImSwitchServer` thread. So if you can talk with the server, then he'll do its stuff and return - without blocking the live viewer. The `runOnUIThread` is used - if I'm not mistaken - by the imscripting module mostly

beniroquai, 2023-06-19:
But what if you need to wait until the stage has moved anywhere? Would you mind sharing your script with me? ;-)

Mine in ImScripting (should use same API stuff) looks like this and after every stage movement (which I implemented in our own library as a blocking operation, because I want to know when I'm "there") leads to a freeze in the GUI (e.g. non responsiveness of buttons and stuff..)

```py
import numpy as np
import time
import threading 
mainWindow.setCurrentModule('imcontrol')

api.imcontrol.enalbeMotors(None, False)
# x,y,z
positions = [	(0, 80000, -4290),
				(17000, 80000, -4340),
				(17000, 96000, -4290),
				(9000, 96000, -4290)]
positions = [(9000, 7000, 110),
				(18000, 16000, 110),
				(54000, 25000, -130),
				(63000, 52000, -240)]

positionerName = api.imcontrol.getPositionerNames()[0]
print(api.imcontrol.getPositionerPositions())
api.imcontrol.setPositionerSpeed(positionerName, "X", 25000)
api.imcontrol.setPositionerSpeed(positionerName, "Y", 25000)
api.imcontrol.setPositionerSpeed(positionerName, "Z", 25000)

api.imcontrol.homeAxis(positionerName, "X")
api.imcontrol.homeAxis(positionerName, "Y")
time.sleep(15)
# move to inital position
# X
position = positions[0]
posX = position[0]
api.imcontrol.movePositioner(positionerName, "X", posX, True, True)
# Y
posY = position[1]
api.imcontrol.movePositioner(positionerName, "Y", posY, True, True)
#Z
posZ = position[2]
api.imcontrol.movePositioner(positionerName, "Z", posZ, True, True)

#
def doScanning():
	iiter = 0
	while 1:
		iPos = 0
		for position in positions:
			print(position)
			posX = position[0]
			api.imcontrol.movePositioner(positionerName, "X", posX, True, True)
			# Y
			posY = position[1]
			api.imcontrol.movePositioner(positionerName, "Y", posY, True, True)
			#Z
			posZ = position[2]
			api.imcontrol.movePositioner(positionerName, "Z", posZ, True, True)

			time.sleep(1)
			
			api.imcontrol.snapImageToPath(str(iPos)+"_posscreening") 
			iPos +=1
		time.sleep(60)
		iiter +=1
		print(iiter)

doScanning()
```

jacopoabramo, 2023-06-21:
@beniroquai sorry for the late reply.

So what you want to do is having a script that does this, correct? Then it would be something like this: 

```python
import numpy as np
import time
import threading

mainWindow.setCurrentModule('imcontrol')

api.imcontrol.enalbeMotors(None, False)
# x,y,z
positions = [	(0, 80000, -4290),
				(17000, 80000, -4340),
				(17000, 96000, -4290),
				(9000, 96000, -4290)]
positions = [(9000, 7000, 110),
				(18000, 16000, 110),
				(54000, 25000, -130),
				(63000, 52000, -240)]

positionerName = api.imcontrol.getPositionerNames()[0]
print(api.imcontrol.getPositionerPositions())
api.imcontrol.setPositionerSpeed(positionerName, "X", 25000)
api.imcontrol.setPositionerSpeed(positionerName, "Y", 25000)
api.imcontrol.setPositionerSpeed(positionerName, "Z", 25000)

api.imcontrol.homeAxis(positionerName, "X")
api.imcontrol.homeAxis(positionerName, "Y")
time.sleep(15)
# move to inital position
# X
position = positions[0]
posX = position[0]
api.imcontrol.movePositioner(positionerName, "X", posX, True, True)
# Y
posY = position[1]
api.imcontrol.movePositioner(positionerName, "Y", posY, True, True)
#Z
posZ = position[2]
api.imcontrol.movePositioner(positionerName, "Z", posZ, True, True)

#
def doScanning():
	iiter = 0
	while 1:
		iPos = 0
		for position in positions:
			print(position)
			posX = position[0]
			api.imcontrol.movePositioner(positionerName, "X", posX, True, True)
			# Y
			posY = position[1]
			api.imcontrol.movePositioner(positionerName, "Y", posY, True, True)
			#Z
			posZ = position[2]
			api.imcontrol.movePositioner(positionerName, "Z", posZ, True, True)

			time.sleep(1)
			
			api.imcontrol.snapImageToPath(str(iPos)+"_posscreening") 
			iPos +=1
		time.sleep(60)
		iiter +=1
		print(iiter)
		
thr = threading.Thread(target=doScanning)

thr.start()
thr.join()
```

I don't think that this will fix the issue of the stuck GUI because `thr.join()` will still probably block the UI. I would maybe reccomend moving the code to the `ImSwitchServer` as I mentioned before. I don't use a script, I do everything at a GUI level.

beniroquai, 2023-06-21:
I'll check. Bit pyro and fastapi use the same thread. No?

jacopoabramo, 2023-06-21:
I don't think it matters that much but frankly I'm not familiar with fastAPI so I really can't say.