ImSwitch/ImSwitch #179: Missing documentation points

jacopoabramo, 2023-06-22:
Desired documentation points are:

1. a list of signals currently included in the ImSwitch architecture:
   - where the signals are originated;
   - their content;
   - their current connections to existing slots;
     - it would be cool (but maybe too time consuming) to make an interactive map of the architecture for people to explore...
2. tutorials on how to add new device managers (or the entire stack, from the model up to the viewer);
    - this also includes how to add signal generators (but this is a bit of a complex topic because we still have to define a general interface not dependent on the nidaqmx package for other hardware DAQ cards);
    - I can take as a reference the implementation of the `PulseStreamManager` but it's not quite yet there unfortunately.
3. tutorial on how to integrate a low level manager;
   - this concept is similar to the [Micro-Manager "hub"](https://micro-manager.org/Building_Micro-Manager_Device_Adapters#complex-device-management-writing-hub-adapters) (take it just as a reference); 
4. tutorial on how to integrate napari plugins with ImSwitch
5. documentation on how to navigate the configuration files more in detail, with reference to the relevant dataclasses that are included within the `ViewSetupInfo`
6. how the fastAPI server works (@beniroquai a run down of this would be nice since you're the one with the most experience)

Point 2, 4 and 5 have been presented in this [workshop](https://www.youtube.com/watch?v=7f1XH7lQcgQ&t=1859s), so I could try to spend some time and commute everything within the documentation. For point 1 I remember I found a package that does interactive presentations or stuff like this and it could become a separate repo within the organization.

@kasasxav I noticed tha the ImSwitch documentation link sends back to your fork of the repo. Can this be fixed somehow? Do I have to rebuild the documentation or change something on the readthedocs website settings?

beniroquai, 2023-06-22:
Great points @jacopoabramo! I would suggest to have a dedicated "hackathon" to work on that with folks at the Leibniz IPHT (worst case just the two of us), but perhaps @jonatanalvelid would be interested to join too. We could block the evening hours and sit down and write down, merge and clean up. What do you think?

beniroquai, 2023-06-22:
I would suggest: First polish then advertise! Could we do that before the
summer break (mine is 18th July ish ;)

> Message ID: ***@***.***>
>

kasasxav, 2023-06-22:
hey @jacopoabramo ,

I just fixed it and also added you as mantainer for the readthedocs, with that you can rebuild the different docs branches. Otherwise they build automatically with each release.

Best,
Xavier.

jonatanalvelid, 2023-06-23:
Hello, I could potentially join and help with a dedicated documentation write-up (at least points 1, 2, 3, 5), depending on when as I will have some early vacation in the beginning of July as well.