ImSwitch/ImSwitch #187: Have a virtual microscope

beniroquai, 2023-09-29:
This is a thread of requirements of a virtual microscope. I have a simple mocker for a stage, illumination and camera. It can move the sample, scale the intensity and defocus it. It has noise too. What else would we need?

https://github.com/ImSwitch/ImSwitch/assets/4345528/56108e99-a17c-4456-aa4f-d076ff757588

jacopoabramo, 2023-09-29:
Awesome work as always :) I'm thinking that this is the perfect spot to also integrate new types of devices such as filter wheels

beniroquai, 2023-09-29:
We could have a virtual sample, too. I'm thinking about a STORM, or even a moving sample. Preparing for the next stage: autonomous microscopy without a microscope ;)