ImSwitch/ImSwitch #188: ImReconstruct - what can it do?

beniroquai, 2023-09-29:
I have never used the ImReconstruct module. What is this really used for? Only reconstructing scanning data?
@kasasxav @jacopoabramo

jacopoabramo, 2023-09-29:
I'm unfortunately not familiar with it. I think @jonatanalvelid also used it, but I may be wrong; I think it was used to do some image processing based on the data acquired from a confocal microscope. It was also mentioned in a paper: https://www.sciencedirect.com/science/article/pii/S246806722300007X

beniroquai, 2023-09-29:
It looks like this:
[image]

Perhaps for STED or Monalisa stuff?

jacopoabramo, 2023-09-30:
> Perhaps for STED or Monalisa stuff?

Yeah I think they use it for those kinds of datasets. I was actually hoping to transform it to do general image processing based on napari plugins and adding the PyImageJ bridge

beniroquai, 2023-10-02:
We could think about a more generic module that may even be capable to bridge @imjoy from @oeway. I specifically thinking about a web-view that renders the ImJoy plugin system. Not sure how we could bridge numpy arrays to automatically be sent to the browser though. The ImBlockly has everything we need already. It shows Blockly in a browser inside Pyqt ;)

jonatanalvelid, 2023-10-02:
It was designed to reconstruct MoNaLISA data specifically (multiple frames with multifoci patterns). Could also be used for multifoci confocal data. I believe there were also some ideas to make it more general by including for example SIM reconstruction as well, however @kasasxav knows better as I was never involved in the development of the reconstruction.
For STED imaging the images are built up directly in the APDManager, no reconstruction necessary.

kasasxav, 2023-10-02:
Hey,

Yes the idea was that it should become something more general but we started with what we had in the lab: MoNaLISA/RESOLFT. The idea is that the reconstruction code itself it's in a DLL file so that implementing for other techniques would only require to load a different file, and maybe also changing the widget where the parameters are shown. The module has different functionality to load the dll, call its functions, etc. In practice I think it became hard to make it general.

In the end I think that people weren't that comfortable having DLLs and want to write the reconstruction code directly in Python so then I agree that the module should be adapted to that :)

beniroquai, 2023-10-04:
Great! Thanks a lot! :)

> Message ID: ***@***.***>
>

closed 2024-04-30 by beniroquai