ImSwitch/ImSwitch #189: FutureWarning for accessing napari qt_viewer

jacopoabramo, 2023-10-14:
Since napari 0.5.0 is approaching release, I think we should focus on the following warning:

```
FutureWarning: Public access to Window.qt_viewer is deprecated and will be removed in
v0.5.0. It is considered an "implementation detail" of the napari
application, not part of the napari viewer model. If your use case
requires access to qt_viewer, please open an issue to discuss.
```

From the stack trace I see that these warnings are caused by the following code in the `NapariBaseWidget` class:

```py
        # Move layer list to bottom
        napariViewer.window._qt_window.removeDockWidget(
            napariViewer.window.qt_viewer.dockLayerList
        )
        napariViewer.window._qt_window.addDockWidget(
            napariViewer.window.qt_viewer.dockLayerList.qt_area,
            napariViewer.window.qt_viewer.dockLayerList
        )
```

@kasasxav do you know what does this do? Is it necessary or can we remove it?

jacopoabramo, 2023-10-14:
Currently the napari layer list looks like this:

[image]

While ImSwitch napari looks like this:

[image]

To me they look ratherly similar. Or am I missing something?

jacopoabramo, 2023-10-14:
I see that this is used as a way to add custom napari widgets to the interface anyway...

beniroquai, 2023-10-17:
Should we do anything about it?

kasasxav, 2023-10-17:
Heyy hope all is good guys :)

I would say try to add an issue to their repo and ask for an alternative way, I used something similar to add available napari plugins to ImSwitch and the code was something they suggested. So probably they have better alternatives now.

jacopoabramo, 2023-10-17:
@kasasxav just to prepare the issue properly, do you know what that code does by any chance?

kasasxav, 2023-10-17:
Sorry forgot to say what it does,
NapariBaseWidget is a general class for dealing with Napari within ImSwitch.

There are some widgets that are subclass of NapariBaseWidget, like the shift buttons or the Reset for update levels (https://github.com/ImSwitch/ImSwitch/blob/b05a431320671c2af16b6db35bed6818d4826d70/imswitch/imcontrol/view/widgets/ImageWidget.py#L19) . The shift buttons are the arrows in the GUI that can align one image to another. In this case, the add and remove dock functions deal with adding a widget inside the napari GUI.

Also, there is a subclass of NapariBaseWidget called NapariHybridWidget, that is used as a superclass for those widgets that want to add stuff into the viewer like the ULensesWidget that adds a layer of an array of dots superimposed to the image. But they don't use specifically the functions addDockWidget and removeDockWidget

Czaki, 2023-10-17:
`qt_vierer` deprecation is postponed to 0.6.0 https://github.com/napari/napari/pull/6283
You still may depend on `Viewer.window._qt_viewer` if you do not plan to support different backends in the future, but then I suggest setup testing against the main branch of the napari repository https://github.com/4DNucleome/PartSeg/blob/develop/.github/workflows/test_napari_repo.yml

As you see, you are a standalone application, not a plugin. 
Also, QtViewer itself is available in `napari.qt` module as part of public API. 

So instead of using Viever, you may use `QtViewer` and `ViewerModel`. 
You need then create own main window, but you will have full control over its design. You may see https://github.com/4DNucleome/PartSeg as an example of an application that provides its own GUI but uses QtViewer for rendering data.